mjg59/python-broadlink #1: device.discover() ?

farhanito, 2016-09-17:
noob question,

device.discover() didn't returns anything. what am i missing?

`osmc@osmc:~$ python
Python 3.4.2 (default, Oct  8 2014, 14:38:51)
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
>>> device = broadlink.rm2()
>>> device.discover()
>>>
`

regards,

mjg59, 2016-09-18:
Do you have a Broadlink rm2 on the same network as the machine you're running discover() from?

farhanito, 2016-09-18:
yes, both on the same network

mjg59, 2016-09-19:
What timezone are you in?

farhanito, 2016-09-19:
GMT+7

mjg59, 2016-09-19:
Ok - the bit that says:

    packet[0x08] = 0xf9
    packet[0x09] = 0xff
    packet[0x0a] = 0xff
    packet[0x0b] = 0xff

Can you change that to:

    packet[0x08] = 0x07
    packet[0x09] = 0x00
    packet[0x0a] = 0x00
    packet[0x0b] = 0x00

and see if it works any better?

farhanito, 2016-09-19:
Ok. Here's what I got.. 
[http://i.imgur.com/sUFSLaf.jpg](http://i.imgur.com/sUFSLaf.jpg)

farhanito, 2016-09-19:
changed to Python2.7, everything seems works now

regards,

mjg59, 2016-09-22:
Oh, cool - I'll fix up the timezone code and try to make it work under Python 3. Thanks for testing!

LinZihong, 2016-10-01:
My time zone is GMT+8 and I changed the bytes to
`packet[0x08] = 0x08
packet[0x09] = 0x00
packet[0x0a] = 0x00
packet[0x0b] = 0x00`
according to my guess, but it didn't work. Am I wrong?

closed 2016-10-15 by mjg59