mjg59/python-broadlink #2: Does this support RF signals?

pruppert, 2016-09-24:
First, thank you. This looks promising.

The RM Pro  2 can also send RF commands. Does this module support that? If not, is it planned?

Thanks.

mjg59, 2016-10-10:
I don't have any RF devices, so haven't been able to test. It's possible that it'll just work with the current code?

aldodemi, 2016-10-10:
I think, and in the next days I'll test it, that you can learn and send also RF code as you can do with IR.

mjg59, 2016-10-25:
Ok, it looks like there's another command for scanning for RF signals. Let me look at the code.

mjg59, 2016-10-25:
Looked some more, and I'm afraid I can't find a specific call for RF learning. With luck the existing IR calls will also work for RF, but I don't have any hardware to test!

jcoudrais, 2016-10-25:
Sorry to talk about rm-bridge.fun2code.de one more time,  but as it is in learning mode,  it can learn both ir or rf (433mhz for sure and perhaps us frequency)

mjg59, 2016-10-26:
@jcoudrais From looking at the code, my expectation is that RF will just work as is. But until someone tests, I have no way of knowing for sure.

jcoudrais, 2016-10-26:
Hi mjg59, i'll do this test.
How could I write this data to be readable ?
because print(ir_packet) gives that 👍 &H(▒6686778676656
PuTTY>>> PuTTY

jcoudrais, 2016-10-26:
OK, that's it:
print str(ir_packet).encode('hex')
I'll test IR

jcoudrais, 2016-10-26:
So, it is ok for the leraning mode and as I send a RF 433 code, it learns that:
e900260109040604150500012c9511131237123712131213121312131114113811141312113812371137123712371213121312131237123712131213121312371237123712131114113811371237120003d700012c9511141139113612141213111411131312123712131213133711371237123713361213121411131236143513121312131311371336123713121214113910371237120003d700012e9213121436123613121312131214111313133513121411133713351336133613361411131213121336143513121312131114371137133514111313123712361337120003d700012d9313121336133613121411131214111312133614111312133613361337133513361312131213121336143513121312131213371236133613121312133613351436130005dc0000

It sends the good code but with differences rm-bridge gives that:
"data":"e900200100012d9412141137113712131214111411141114113712131213123712371237123712371213121312131138113811141115101411371237123713121213123712371237130003d600012c9511151038113712151014111510131213123712141115103712371237123712381113131212131237123712131213121312361337123712131212133613361336130003d700012c9413121336133613121312131213121213123712131312123712371336133613361312121411131336123712131214111312361336153413121511123613381136130003d700012e9214111435143711121314111213141112133613121411143513371237123613361411131213121435133613131212131213361336143513121312133613361435130005dc0000000000000000"

mjg59, 2016-10-26:
The exact code received may well vary based on conditions. If the learned code works when you play it back, I think that's working fine!

mjg59, 2016-10-30:
Closing since this appears to work with the existing code.

closed 2016-10-30 by mjg59