mjg59/python-broadlink #3: device = broadlink.rm2() results in socket.error: [Errno 49] Can't assign requested address

pruppert, 2016-09-25:
I run the following on Mac

```
import broadlink

device = broadlink.rm2()
```

and get the error:

```
  File "/Library/Python/2.7/site-packages/broadlink-0.1-py2.7.egg/broadlink/__init__.py", line 15, in __init__
    s.connect(('8.8.8.8', 0))  # connecting to a UDP address doesn't send packets
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 49] Can't assign requested address
```

Any idea what the problem is?

LinZihong, 2016-09-30:
I encounter the same problem

eddyyanto, 2016-10-01:
It seems like the socket failed to bind on default special address 0.0.0.0. Maybe try changing it to 127.0.0.1 instead?
`self.id = bytearray([0, 0, 0, 0])`
to
`self.id = bytearray([127, 0, 0, 1])`

eddyyanto, 2016-10-01:
Sorry, please ignore or delete my earlier comment, I mistook id for ip address.

LinZihong, 2016-10-01:
I resolved this problem by deleting the `s.connect` line and replaced the next line with your computer's ip in your local network(e.g. `local_ip_address="192.168.1.123"`)

kelvl, 2016-11-03:
Looks like on macOS, it requires a port to "connect" to. I am guessing that we want to connect against Google's DNS, adding port 53 fixed the issue for me. File a PR at #11

PR #11 (2016-11-04): Fix for [Errno 49] Can't assign requested address
Fixes #3 on macOS
commit bd1c4448 (2016-11-04): Fix for [Errno 49] Can't assign requested address

Fixes #3 on macOS
closed 2016-11-04 by mjg59