mjg59/python-broadlink #4: Need some help, please

aldodemi, 2016-09-29:
First of al,l many thanks for your work!
All work great but I found same little problem.....
First time I do device.auth() there is an error
>>> device.auth()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "broadlink/__init__.py", line 98, in auth
    payload = aes.decrypt(str(response[0x38:]))
  File "/usr/lib/python2.7/dist-packages/Crypto/Cipher/blockalgo.py", line 295, in decrypt
    return self._cipher.decrypt(ciphertext)
ValueError: Input strings must be a multiple of 16 in length
Second time I do device.auth() all seem ok.....

>>> device.auth()

When I do device.enter_learning(), RM2 enter correctly in learning mode, and this is ok, but as I'not more a programmer and don't know "python" I need some help to record learned codes and how to resend those codes.

Thank you in advance for everything you can do to help me :-)

Aldo

aldodemi, 2016-09-29:
My goal is to use your great works in "Domoticz".
Aldo

eddyyanto, 2016-10-01:
I tried it on my system (Win 8 + Python 2.7.5) and it works. Here's how I send the code:

`import broadlink
import time

device = broadlink.rm2()
device.discover()

device.auth()
device.enter_learning()
time.sleep(2)
ir_packet = device.check_data()

 # record learned code to file
f = open('on.ir','w')
f.write(ir_packet) # python will convert \n to os.linesep
f.close() # you can omit in most cases as the destructor will call it

time.sleep(2)
device.send_data(ir_packet)`

mjg59, 2016-10-30:
Closing since there seems to be a good example for this, thanks!

closed 2016-10-30 by mjg59