mjg59/python-broadlink #5: Support for other broadlink devices such as AirPlug and SmartPlug

eddyyanto, 2016-10-03:
Hi @mjg59 ,

Thank you for the API! I tested the discover() method with 2 other Broadlink devices and it is able to detect both AirPlug and SmartPlug. Of course, unlike the RM2 IR Controller, these 2 devices are built for different purposes and do not have the learning mode. At its core:
- AirPlug generates read-only values such as humidity, temperature, noise level and luminance. These values can be read using official app.
- SmartPlug has a boolean values which can be read and set using the official app.

Do you have any plan to add support for other Broadlink devices including the above so we can get and change its internal state? I understand the original API is only meant for the RM2 IR Controller. Anyway, I have attached[ 2 wireshark capture files](https://github.com/eddyyanto/python-broadlink/blob/master/airplug-smartplug-pcap.zip) if you want to see it.

**192.168.0.129**	: Broadlink App
**192.168.0.161**	: Sstem running Chrales Proxy and Wireshark

eddyyanto, 2016-10-03:
By the way, I am interested to know what tools you're using to analyze the packets between RM2 and the Broadlink app and what method did you use to determine the protocol and encryption.

I am using Charles Proxy as a middleman for my phone in which the Broadlink app is installed. I also ran Wireshark on the same system as the Charles. Not sure how ideal my setup is though, I am new to this.

aldodemi, 2016-10-03:
HI @eddyyanto , I'm also interested in read AirPlug information (I have one), how did you manage to discover it?

TerranUlm, 2016-10-03:
I've got my Broadlink SP2mini "SmartPlug" a few days ago and tried to find a way to integrate it to openhab or homegear. The best "ready to use" solution so far seams to use an app for android or iphone which acts as a bridge. For android I've found "Broadlink RM Tasker Plugin" and "RM Bridge". Both act as HTTP bridges to the broadlink API. RM Bridge seams limited to the RM device only, Broadlink RM Tasker Plugin is able to control my SP2mini.
But I'm not happy with that solution as it adds an extra layer for errors.

So I'd searched githup for "broadlink" and found a few interessting projects here. One of them is a refactored version of the original SDK example (https://github.com/torinnguyen/broadlink-android) which gives a lot of information of the original API, but sadly not about the actual API code itself which is closed source. But that way it's a lot easier to identify the protocol itself, which is JSON based.

Also a russian guy (foxikmax) has created a simple "replay attack" php suite which can control an airconditioner (https://github.com/foxikmax/airHome-PHP). That was the first time I've seen a tcpdump of the transmitted packages. He states that he identifed the packages by the 5aa5aa555aa5aa55 sequence. As I understood he captured them on his wifi router, so he actually got them by a man-in-the-middle attack.

Then I've looked at this project here and was amazed as Matthew somehow figured the encyption part of the protocol!

So I tried to capture the communication myself, but with my current network setup that is not as easy as I cannot do a men-in-the-middle attack on the wifi router. So I used arpspoof to redirect packages intended for the device to my raspi, so I could capture at least that part of the communication (https://github.com/TerranUlm/python-broadlink/tree/reverse-engineering/pcaps).

What I've noted so far is that the send_packet function of the python library assumes that packet[0x24] is always 0x2a - but that is different on my capture and different on foxikmax's capture.
Also the command varies in my captures, 4 times it is "65" and not "6a". I'd switched the state 4 times during my capturing.

Btw: I've used wireshark 2.2.0 and it identifies the protocol as "QUIC" (Quick UDP Internet Connections). The 5aa5aa555aa5aa5500 part seams to be the QUIC header. QUIC is a UDP implementation of Googles SPDY protocol which requires encryption (https://en.wikipedia.org/wiki/QUIC).  So maybe that is another way of approaching this. :8ball: But that is just a guess.

TerranUlm, 2016-10-03:
After some more analyzing I'll think the "65" command is something else, all the packets have the same content.

aldodemi, 2016-10-04:
You can find some capture file from my access point here: https://github.com/aldodemi/for-broadlink see AP-CUCINA files recorded when the smartphone ask for information to the A1 AirPlug.

eddyyanto, 2016-10-04:
> HI @eddyyanto , I'm also interested in read AirPlug information (I have one), how did you manage to discover it?

@aldodemi — make sure only one Broadlink device is in the network and use the python API to discover it.

`import broadlink
import time

device = broadlink.rm2()
device.discover()

print device.host`

aldodemi, 2016-10-04:
Thank @eddyyanto , as I've both, an RM2 and an A1 when I do a device.discover I found sometime one and sometime the other one.........

TerranUlm, 2016-10-04:
I've figured how to capture the packages on my old aged wlan AP - not that easy to read but all the necessary data is there. I've also extracted the data itself (but missed the last 4 bytes of every package, see readme.txt (https://github.com/TerranUlm/python-broadlink/tree/reverse-engineering/pcaps)).

Time to go to bed...

mjg59, 2016-10-25:
Latest commit adds support for the A1, I'll need to look a little more into the SP2.

aldodemi, 2016-10-25:
Need more help, how can I do a device.discover() for the A1?
Many thanks.

mjg59, 2016-10-25:
Just use the same function - I need to clean that up. If you have multiple devices on the same network it'll be random which one you get right now.

aldodemi, 2016-10-25:
I know so I've done a test as my two devices have fixed IP, but when I'm sure to select A1 to check sensors there was an error:
 temp_packet = device.sensors()
AttributeError: rm2 instance has no attribute 'sensors'

this is my code:
#!/usr/bin/python
#
import broadlink
import time
device = broadlink.rm2()
device.discover()
print device.host
#
# .15 RM2
# .16 AIR A1
#
while '192.168.10.15' in device.host[0]:
        print device.host
        device.discover()
else:
        try:
                device.auth()
        except ValueError:
                device.auth()
        temp_packet = device.sensors()
        print temp_packet

can you help me to understand my error?
Thank

eddyyanto, 2016-10-26:
> Latest commit adds support for the A1, I'll need to look a little more into the SP2.

Many thanks, @mjg59 ! It works with my test A1 device.

@aldodemi , I don't think the library can iterate through a list of connected broadlink devices yet. Your best bet is to disconnect the other devices except A1. Then use the library like this:
```
import broadlink
import time

device = broadlink.rm2()
device.discover()
device.auth()
print device.check_sensors()

```
Here's the reply from my A1:

`{'air_quality': 'good', 'light': 'normal', 'noise': 'normal', 'temperature': 30.8, 'humidity': 66.3}`

mjg59, 2016-10-26:
@aldodemi The command is check_sensors(), not sensors()

mjg59, 2016-10-26:
@eddyyanto I've now got both devices hooked up on the same network segment, so I'll add support for enumerating a list of devices shortly

mjg59, 2016-10-30:
Current code has support for multiple devices - check the README for updated instructions on this.

closed 2016-10-30 by mjg59

eddyyanto, 2016-10-31:
Thank you, @mjg59!

markvovo, 2016-11-14:
Does anyone else see other data packets in their wireshark capture in the QUIC stream??
I am seeing QUIC tagged packets from my Router, to my Laptop IP address, being sent to Japan Amazon IP address, along with the 1 message to my RM Pro.  So, out of say 10 messages in the QUIC, 9 are about my laptop traffic, and 1 is to port 80 on external Server.  Note: my laptop has nothing to do with any communications with my RM Pro - I use the econtrol app on my iPhone , and capture turning on/off my air conditioner .  Should I be worried about this "included" data?????

mjg59, 2016-11-21:
Wireshark will flag anything sent to UDP port 80 as QUIC, even though most of it isn't actually QUIC. Multiple different applications may be doing this.

markvovo, 2016-11-21:
Thanks for replying @mjg59 , but the 1 packet, has multiple sub packets.  It is NOT different packets, with data on UDP going to port 80.  
I understand what you are trying to say, but this appears not to be the case here.
I will do another wireshark capture, and post when I can, to show what exactly I mean, later this week.

kassyok, 2017-05-06:
Hi guys, is there an easy way for a non experienced user like me to create a home screen widget that displays the Broadlink A1 values?

If yes, where should i start?

mob41, 2017-05-06:
@kassyok The very first thing to say is: <b>You are off-topic in this issue.</b> You should or even must to create an issue about your question.