mjg59/python-broadlink #6: temperature

lepipo, 2016-10-12:
Hi, 
first, i'll thanks to you script, it's amazing it's work with my (new) broadlink RM-PRO and on raspberry pi2! 

I'll ask you something, can we get the temperature sensor value with this script? i'll plan to integrated it on jeedom or domoticz.

Regards

mjg59, 2016-10-15:
Does the RM-PRO have its own temperature sensor? If so, I'll try to add support.

lepipo, 2016-10-16:
Yes, the RM-PRO have its own temperatur sensor. It will be very cool if you can add this support. In fact, i'll can schedule to start my air conditioner if the temperature is up to.. i hope you can find the parameter !
many thanks

mjg59, 2016-10-25:
Added in the latest commit. Thanks!

closed 2016-10-25 by mjg59

lepipo, 2016-10-25:
Hi, i'm newby on python, i see your change in your script, but i don't know how can i get the temperature value ! i do print device.check_temperature() but i have always the same answer : "none" what's wrong?

aldodemi, 2016-10-25:
With "device.check_temperature()" I've a return value but, let me say, is "strange": 120.8 from RM2 and 168.3 from A1, not celsius but not also Fahrenheit.....

jcoudrais, 2016-10-25:
For me, it's ok, I get 25.1 also °C

aldodemi, 2016-10-25:
@jcoudrais , can you please share your script to retrieve the temperature? Thanks

jcoudrais, 2016-10-25:
Here : 
import broadlink
device = broadlink.rm2()
device.discover()
device.auth()
print device.check_temperature()
25.1

aldodemi, 2016-10-25:
like mine, except for the result: 120.8!!!!!
Thank you.....