mjg59/python-broadlink #9: ascii reading code

jcoudrais, 2016-10-25:
Thank you for your job, I plan to use it on my very little nas, that is better than rm-bridge with an old android phone wifi connected...

Should we get ascii codes in the same format than http://rm-bridge.fun2code.de/, because it's a very good way to view the learned codes and also backup them...

Also, should we get them into a file to just send a code like "ampli_off", like rm-bridge does ?
 Then a py file to manage that send codes user friendly...

{"data":"260058000001269114121213121312131213121312131236141212131312121313361336133713121336133713361336143613131212133614121312131113131312123713361313120005a30001264913000c400001264913000d05","mac":"b4:43:0d:38:72:d9","name":"ampli_off"}

mjg59, 2016-10-25:
This code is intended to be the low level interface, allowing people to build tools that handle the user interface in a nicer way. I think it makes more sense to put the code parsing in those tools rather than directly in here.

jcoudrais, 2016-10-25:
You're right, and you did the job! Thanks. 
 I need to become a developer again 😉

closed 2016-10-25 by jcoudrais