mjg59/python-broadlink #10: [not an issue] this works with RM3 Mini a.k.a Blackbean

farhanito, 2016-11-02:
I just tested and it works with RM3 Mini. 

Thanks a lot for this

prodigy7, 2016-11-08:
Hi farhanito,

can you tell me how you get it working? I've tried it with`import broadlink

devices = broadlink.discover(timeout=5)
devices[0].auth()
devices[0].host
devices[0].enter_learning()
ir_packet = devices[0].check_data()
print(ir_packet);`
But get nothing. Maybe you could post sample code which should work?

farhanito, 2016-11-09:
No errors? 
Which python version do you use?

prodigy7, 2016-11-09:
No errors. Does simply not receive anything. Device is detected but key press not.
Is my sample right? Is it maybe necessary to use check_data() inside a loop or something like that? Generellay it would be maybe help, if a working test script would be included in git repo.

mjg59, 2016-11-11:
@prodigy7 check_data will return nothing until the device has something to return, so a loop is probably necessary.

prodigy7, 2016-11-11:
Could you support me with a quick and dirty sample? Would be nice. I've skills in php, perl, js, vb.net .... but python is currently not in my skill list.

tyjtyj, 2016-11-12:
try this 

devices = broadlink.discover(timeout=5)
devices[0].auth()
print "enter_learning"
device.enter_learning()
time.sleep(5) #you need to press your remote in 5 sec, can adjust for longer time.
ir_packet = device.check_data()
print str(ir_packet).encode('hex')

prodigy7, 2016-11-12:
Coooooool! It works - one step closer integrate the device in my home automation (www.fhem.de). Thank you very much! I got it working with following sample:
```python
#!/usr/bin/python

import broadlink
import time

print "discover"
devices = broadlink.discover(timeout=5)
devices[0].auth()
print "enter_learning"
devices[0].enter_learning()
print "sleep"
time.sleep(5)
print "check data"
ir_packet = devices[0].check_data()
print str(ir_packet).encode('hex')
devices[0].send_data(ir_packet)
print "wait"
time.sleep(2)
devices[0].send_data(ir_packet)
```

Some how, only the first time the received code works. Second time not. Any idea why?

Another question: The lirc project has a really big data base with code for different devices.  Do you think it's possible could use them somehow? Maybe with a function convert that codes to a for that device understandable code?

tyjtyj, 2016-11-12:
I can help with this
Maybe with a function convert that codes to a for that device understandable code?
print str(ir_packet).encode('hex') will give you some "b212323546756.......768"

used the function below to send the data

```
def senddata(hex_data):
	device.send_data(hex_data.decode('hex'))
	return

eg.
senddata("b212323546756.......768")
```

Sorry I am not python programmer. There might be better code but this works for me.
U should start new issue as this is to discuss RM3 specific.

PS: I wonder will @mjg59 make this python-broadlink compatible with ulfire for amazon echo.

prodigy7, 2016-11-12:
The following hex code was the result of pressing the VOL UP on my LG Remote.

First time:
```2600 5800 0001 2894 1015 1015 103a 1014 1015 1015 1015 1015 1039 1139 1015 103a 1039 1139 103a 103a 1014 103a 1015 1015 1015 1015 1014 1015 103a 1015 1039 1139 103a 103a 1039 103a 1000 0529 0001 2949 1100 0c5a 0001 284a 1000 0d05```

Second time:
```2600 5000 0001 2894 1015 1015 1039 1114 1015 1015 1015 1015 1039 1139 1015 103a 1039 1139 103a 103a 1014 103a 1015 1015 1015 1015 1014 1015 103a 1015 1039 1139 103a 103a 1039 103a 1000 0529 0001 294a 1000 0d05 0000 0000 0000 0000```

The corresponding lirc file could be http://lirc.sourceforge.net/remotes/lg/AKB72915207
(I try to get my old IR working under linux with lirc so I get the code from my remote directly)

mjg59, 2016-11-12:
@prodigy7 Make sure you're running the latest version of the code - there's a fixed bug that would prevent repeated commands from working.

prodigy7, 2016-11-12:
Right! Pulled current version and now it works. By the way: On my RM mini I can get the name set for the device with
```
name = responsepacket[0x40:0x54]
```
Maybe on other devices it works too!?

prodigy7, 2016-11-18:
I found http://irdb.tk/codes/ - is it possible make the codes there some how usable?

eddyyanto, 2016-11-19:
@prodigy7 I just looked at [irdb.tk](http://irdb.tk/) and I think technically you can copy the **UEI Hex** (**not Pronto Hex**) to be used with python-broadlink. For example, copy the UEI Hex and format it like below before you send it off to the device:

```
ir_packet = bytearray([0x00 ,0x00 ,0x00 ,0x00 ,0x00 ,0x00 ,0x08 ,0x00 ,0x00 ,0x00 ,0xe0 ,0x07 ,0x25 ,0x32 ,0x07 ,0x01 ,0x03 ,0x0a ,0x00 ,0x00 ,0x00 ,0x00 ,0xc0 ,0xa8 ,0x00 ,0x81 ,0x53 ,0xf1 ,0x00 ,0x00 ,0x3d ,0xc3 ,0x00 ,0x00 ,0x00 ,0x00 ,0x06 ,0x00 ,0x00 ,0x00 ,0x00 ,0x00 ,0x00 ,0x00 ,0x00 ,0x00])

device.send_data(ir_packet)
```

prodigy7, 2016-11-19:
I try it but something goes wrong:
```python
#!/usr/bin/python

import broadlink
import time

print "discover"
devices = broadlink.discover(timeout=5)
devices[0].auth()

print "enter_learning"
devices[0].enter_learning()

print "send data"
ir_packet = bytearray([0x00, 0x00, 0x2F, 0x00, 0xD0, 0x06, 0x11, 0xA0, 0x08, 0xD0, 0x01, 0x1A, 0x03, 0x4E, 0x01, 0x1A, 0x01, 0x1A, 0x01, 0x1A, 0x5A, 0xDE, 0x11, 0xA0, 0x04, 0x68, 0x01, 0x1A, 0xBB, 0xCE, 0x22, 0x01, 0x22, 0x22, 0x22, 0x21, 0x22, 0x22, 0x22, 0x22, 0x12, 0x12, 0x22, 0x21, 0x21, 0x21, 0x11, 0x13, 0x82, 0x45])
devices[0].send_data(ir_packet)
```
Original UEI Hex was
```
00 00 2F 00 D0 06 11 A0 08 D0 01 1A 03 4E 01 1A 01 1A 01 1A 5A DE 11 A0 04 68 01 1A BB CE 22 01 22 22 22 21 22 22 22 22 12 12 22 21 21 21 11 13 82 45
```
Error is
```
Traceback (most recent call last):
  File "test5.py", line 15, in <module>
    devices[0].send_data(ir_packet)
  File "/root/python-broadlink/broadlink/__init__.py", line 186, in send_data
    self.send_packet(0x6a, packet)
  File "/root/python-broadlink/broadlink/__init__.py", line 164, in send_packet
    payload = aes.encrypt(str(payload))
  File "/usr/lib/python2.7/dist-packages/Crypto/Cipher/blockalgo.py", line 244, in encrypt
    return self._cipher.encrypt(plaintext)
ValueError: Input strings must be a multiple of 16 in length
```

eddyyanto, 2016-11-20:
There was a recent [merged pull request](https://github.com/mjg59/python-broadlink/pull/16) that fixes this value error issue. Please update your copy and try again.

prodigy7, 2016-11-20:
Pulled latest version but get still the same error:
```
Traceback (most recent call last):                                                                                                                                                                             
  File "test5.py", line 17, in <module>                                                                                                                                                                        
    devices[0].send_data(ir_packet)                                                                                                                                                                            
  File "/root/python-broadlink/broadlink/__init__.py", line 327, in send_data                                                                                                                                  
    self.send_packet(0x6a, packet)                                                                                                                                                                             
  File "/root/python-broadlink/broadlink/__init__.py", line 209, in send_packet                                                                                                                                
    payload = aes.encrypt(str(payload))                                                                                                                                                                        
  File "/usr/lib/python2.7/dist-packages/Crypto/Cipher/blockalgo.py", line 244, in encrypt                                                                                                                     
    return self._cipher.encrypt(plaintext)                                                                                                                                                                     
ValueError: Input strings must be a multiple of 16 in length
```

keptenkurk, 2016-11-20:
@prodigy7 The UEI hex ir_packets on irdb.tk are 40 byte long.
Python-broadlink adds 4 byte before encrypting. To become a multiple of 16, the packet need to be 4 byte longer. Padding a Samsung packet with 0's didn't show errors anymore but neither a response on the TV.

mjg59, 2016-11-20:
The UEI format isn't the same as the Broadlink format. I have the Broadlink format partially decoded, but not entirely - the majority of the packets are simply lengths of signal, alternating between on and off. But I don't have an explanation for the lead-in bytes yet.

tyjtyj, 2016-11-27:
@prodigy7,

Do you have broadlink hex code comparable to UEI hex ?

i notice all ir 1st 4 char is 2600. 

eg turn on lg tv 2015 model
260058000001289313121212143513121411131114111411133614361311133613361336143514361312131212371311141114111435133614351435141212371336133614101411140005260001264913000c4f0001274814000d05

Also this senddatahex will zero fill to fix this error
"Input strings must be a multiple of 16 in length"

```
def senddatahex( hex_data ):
	#count zero padding
	pad_len = 32 - (len(hex_data) - 24) % 32
	hex_data = hex_data + "".ljust(pad_len, '0')
	device[0].send_data(hex_data.decode('hex'))
	return

```

tyjtyj, 2016-12-10:
@prodigy7, I managed to partially reverse engineer http://lirc.sourceforge.net/remotes/lg/AKB72915207 to broadlink compatible code 

Consider your example 
First time:
``
2600 5800 0001 2894 1015 1015 103a 1014 1015 1015 1015 1015 1039 1139 1015 103a 1039 1139 103a 103a 1014 103a 1015 1015 1015 1015 1014 1015 103a 1015 1039 1139 103a 103a 1039 103a 1000 0529 0001 2949 1100 0c5a 0001 284a 1000 0d05
``

Split to several section below
``
2600 - Broadlink IR code header
``
``
5800 0001 2894 - LG Header(I dont know how to produce this yet )
``

and the 16 bit data here
``
1015 1015 103a 1014  1015 1015 1015 1015  1039 1139 1015 103a  1039 1139 103a 103a  - Remote header
``
Consider 0 = 1015, 1 = 103a/9 and Translate to 16 bit below. 
``
0010 0000 1101 1111
``
Data from your link say 'pre_data       0x20DF'. Where 20DF to binary is same as above data

The second 16 bit the remote code. Given Vol up is 0x40BF. Where 40BF to binary you will get 
``
0100 0000 1011 1111
``
Which represet by broadlink code below
``
1014 103a 1015 1015    1015 1015 1014 1015   103a 1015 1039 1139   103a 103a 1039 103a    
``
The remaining information is the tail which i dont know how decode
``
1000 0529 0001 2949 1100 0c5a 0001 284a 1000 0d05
``

as long as we generate the broadlink code this way it should work. Keep changing the remote code to represent each button.

[Broadlink IR code header] + [LG Header] + [remote code. eg 40BF] + [tail]

tobiaswaldvogel, 2017-01-15:
@tyjtyj  As my RM2 Pro is on its way and I plan to use it from my router in some "C" code  I read through this.
Thanks for the analysis you have done so far. Here are some thoughts from my side:

Having a look at the LIRC definition it seems that the bytes are just a sequence of pulses and space, as @mjg59 already said. So 10 15 means a pulse of  0x10 units and a space of 0x15

As IR is usually using a 38khz carrier I would assume that the unit would be the period length 26.3us (1/38khz), which seems to match roughly.

The first group `5800 0001 2894` means probably:

`5800`  pulse of 0x58 units no space
`0001`  as the sum of 0x58 + 0x28 = 0x80 looks way too short for the header I would assume that his
            has a special meaning of pulse with length 0x100 and no space
`2894`  pulse of 0x28 units, space 0x94

0x58 + 0x100 + 0x28 = 0x180,  0x180 * 26.3 = 10100, which could be the header
0x94 * 26.3 =3892, which looks like the space after the header

I will do some further analysis once my device arrived next week.

PianSom, 2017-04-05:
Has anyone made any more progress with decoding?

I have a Panasonic TV, which needs a long (ie transmit 4 presses) button press to turn on. It only requires one press to turn off. 

If anyone has managed to decode or is able to help, a single press is 
2600cc0072390f0d0e2c0f0d0e0f0e0f0f0d0f0e0e0e100d0f0e0f0d0f0e0f0e0f2a0e0f0f0d0f0e0f0e0e0e0f0e0f0e0f0d0f0e0f2a0e0f0f0d0f0e0f0e0f0d0f0e0f0e0e0e0f2b0e0e0f2b0f2a0f2a0f2a100d0e0f0f2a0f0e0e2b0f2a0e2c0e2b0e0f0e2b0e00099b713a0e0f0f2a0f0e0f0d0e0f0f0d100d0e0f0e0e0f0e0e0f0f0d0e0f0f2a0f0e0e0f0f0d0e0f0e0f0e0e0e0f0e0f0f0d0f2a100d0e0f0f0d0f0e0e0f0e0e0f0e0e0f0e2b0f0e0f2a0f2a0f2a102a0e0e0f0e0f2a0f0e0f2a0f2b0f2a0f2a0f0e0f2a0f000d05000000000000000000000000

and I am trying to work out what four presses would be.

mindigmarton, 2017-04-21:
I managed to create broadlink compatible codes from rc5x pronto hex codes:

Lets take the rc5x 29 63 11 (optical 2 input) code:
0000 0073 (38khz)
0000 0010  (15 pairs + 1 pair lead out)
0020 0020  0040 0040   0020 0020   0020 0020   0040 0040   0020 00A0   0020 0020   0020 0020   0020 0020   0020 0020   0020 0020   0040 0020   0020 0040   0040 0040  0020 0020   0020 0AA8

so I captured the optical 2 IR code with the broadlink, then for simplicity I normalized the values so I can see the patterns.
Fortunately IR receivers accepts imperfect codes so we can normalize the values to whatever we feel is the 'closest'. 

So with this we see that there is almost a 1:1 mapping...
The first 3 words aren't used, instead you have:
2600 (broadlink header)
2200 (length header) which is 34 dec, 15 pairs + 1 pair lead out + 1 unknown before the lead out
0020 becomes 1c (32 original vs 28)
0040 becomes 3c (64 original vs 60)
00A0 becomes 91 (160 original vs 145)
Unknown pair before the lead out is 0f 00.
0AA8 becomes 0d 05 or 00 d5 for repeats (2728 original vs 3333)
Then you need to pad with zeroes to multiples of 16.

The lead out word is probably not necessary if the command does not repeat and the unknown pair can usually be removed without issues.

With this I managed to reproduce discrete power OFF and ON from pronto hexes (http://irdb.tk/encode/), although the ON code doesn't always work which probably has something to do with my wrong normalization.

So far I couldn't figure out why the values are skewed like they are. They are very close to the original values and around the same scale (~1x, 2x...) 
I'm probably missing some math (frequency, delays) which would explain it...

pavram, 2017-04-24:
So; I am just building on the posts of tyjtyj and tobias above.  With my own insight. 

Lets imagine where this data came from.  It is an analog IR receiver being fed into some kind of digital values.  This explains the minor variations into the brightness values (low/high ie xx2A vs xx29 etc.), and slight variations into the timing lengths.  (0Fxx vs 0Exx) 

The pre-amble values (most clearly represented in PianSom's data, the "7239" or "713a" values are also similar in an "analog" way.  They are so close it is silly not to assume that they represent the same underlying value.  tyjtyj's codes also are all so close (0001 2949, 0001 284a, 0001 2894) to not assume that the source value is fundamentally the same (but also, something determined by the nature of the transmission, perhaps it represents the delay between low-high signals?)

It is important to note that looking at tyjtyj's code it seems to end in a weird way that looks like it is transmitting 2 lots of "empty" or "nothing".  But breaking it down I think a command looks like this:

2600
Packet Length
---- Packet start ----
[Some kind of IR deduced value, 16-32 bits (tyjtyj has 32 bit values, PianSom has 16bits)]
[IR Low/High signals] (Format: Duration:Brightness)
[End of IR] (Format: Duration:00)
Finishing with 0d0f if it is end of packet. or some other value if it continues (PianSom has 099b for a 1, tyjtyj has 0529 and 0c5a)
---- Go to Packet Start if the packet continues ---

mindigmarton, 2017-04-24:
Looks like there are a bunch of issues about IF / RF packets. Perhaps they should be closed and merged:
Issue #57 
Issue #27 
Issue #10

This script works perfectly for rc5x pronto ir. For some codes (turn on / off) I had to set repeat to 1, which would explain why I couldn't get them to work reliably before...

https://gist.github.com/appden/42d5272bf128125b019c45bc2ed3311f

closed 2021-02-17 by felipediel