mjg59/python-broadlink #13: how to assign mac address and ip

tyjtyj, 2016-11-07:
Anyway can speed up discovery by assigning the mac address and ip address manually. 

Just 1 device with me and there is no need to discover each time as the mac and ip address always the same. 

I am not python expert. Any example would be great.

264768502, 2016-11-07:
device = broadlink.device(host=host, mac=mac)
you need convert mac string to bytearray

mjg59, 2016-11-08:
device = broadlink.device(host="192.168.1.1", mac=bytearray([0x01, 0x02, 0x03, 0x04, 0x05, 0x07]))

should work, for example.

illxi, 2016-11-08:
device = broadlink.device(host=("your ip", 80), mac=bytearray([0x01, 0x02, 0x03, 0x04, 0x05, 0x07]))

mac is in reverse order.

jcoudrais, 2016-11-08:
for me, mac address is in the right order... 

But I get freezes on python randomely... Any idea ?

tyjtyj, 2016-11-08:
Test this works 
device = broadlink.device(host=("192.168.1.1",80), mac=bytearray.fromhex("aa bb cc dd ee ff"))

Mac does not need to be reverse. Thanks all for help

closed 2016-11-10 by mjg59

tyjtyj, 2016-11-21:
This stopped working after recent update

This works 
devices = broadlink.discover(1)
device = devices[0]

But this does not work anymore
device = broadlink.device(host=("192.168.1.10",80), mac=bytearray.fromhex("b4 43 0d d1 00 00"))

mjg59, 2016-11-21:
Yes, sorry, the move to properly subclassing different device types means that the generic device class is no longer useful - you'll need to express the type by creating the appropriate class.