mjg59/python-broadlink #17: How do you save the command (e.g.,  ir_packet) for use in a script?

pruppert, 2016-11-19:
Could you flesh out the readme a little more describe how to later use the learned command? Do you have to write/read from a file or something? Could you provide some example? Thanks.

keptenkurk, 2016-11-19:
The examples posted [here](http://www.domoticz.com/forum/viewtopic.php?f=17&t=6929&p=104097&hilit=broadlink#p103517) work fine (both ir and rf)

rupino, 2016-11-19:
Thanks. I'll try that.

pruppert, 2016-11-19:
Oops posted from different account. Thanks. I will try.

illxi, 2016-11-20:
Is it possible to load config from the "share to other phones in WLAN" from broadlink's E-control app? It will made setup much easier.

technicalpickles, 2016-12-05:
Another approach is to dump the packet to YAML:

```python
import broadlink
from IPython import embed
import time
import yaml

devices = broadlink.discover(timeout=5)
device = devices[0]
device.auth()
device.enter_learning()

packet = None

print("Learning", end="", flush=True)
while packet is None:
    print(".", end="", flush=True)
    time.sleep(5)
    packet = device.check_data()
print()

print(yaml.dump(packet))
```

I'm taking that snippet, and putting into a YAML like:

```yaml
christmas_lights:
  on: !!binary |
    sgD8AAYSEgYGEgYTBRMRBwUTEQcFEwUTBhISBxEGEgcFExEHBRMFEwUTBRMGEgYSEgYSBwW7BhIS
    BgYTBRMFExEHBRMRBwUTBhIGEhIHEQcRBwUTEQcGEgYSBhIGEgYSBhMRBxEHBbsGEhIGBhMFEwUT
    EQcFExIGBRMGEwUTEQYSBxEHBRMRBwUTBRMGEgYSBhIGExEHEQcFuwYTEQcFEwUTBRMRBwUTEQcG
    EgUTBhISBxEHEQcFExEHBRMFEwYSBRMGEgYTEQcRBwW7BhISBwUTBRMFExEHBRMRBwUTBRMFExIG
    EgcRBwUTEQcFEwUTBRMFEwUTBRMGEgYTBQAF3AAAAAAAAAAAAAAAAA==
  off: !!binary |
    sggyAAcTEgcGEwYTBhMSBwYTEgcGEwYTBhMSBxIHEgcHExIHBhMGEwYTBhMSBxMHBhMGEwa8AAAA
    AAAA
```

And then sending like:

```
import broadlink
import yaml
with open('codes.yaml', 'r') as file:
    data = yaml.load(file)
packet = data['christmas_lights'][True]  # YAML treats on as True
devices = broadlink.discover(timeout=5)
device = devices[0]
device.auth()
device.send_data(packet)
```

upstairs-at-erics, 2017-01-06:
Hi there.
This looks like an excellent integration. I've just placed an order for the RM2 but I can't work out how to install the library for use in Python.
Can anyone help me please - apologies if I'm being dumb :-)

finity69x2, 2018-02-03:
@keptenkurk 

> The examples posted here work fine (both ir and rf)

From the link it looks as tho it only works for IR. How does it need to be modified to work with RF?

And can the HEX codes that are read into the text file be used in other applications?

I've been trying to figure out the codes including protocol, pulse length etc for use in a MQTT gateway using RCSwitch. But RCSwitch doesn't read my remote but the RM Pro has.