mjg59/python-broadlink #21: [request] How can i intercept command from Broadlink TC2

alesms, 2016-12-01:
Hi,
all works great! 
But...now i bought a TC2 and i would get the command sent from my RM PRO to the TC2.
Normally afer you call
devices = broadlink.discover(timeout=5)
devices[0].auth()
you call 
devices[0].enter_learning()
(or you can also don't call this and simply enter in learning mode with app)
and after you sent RF command call
ir_packet = devices[0].check_data()
and, after, you can save ir_packet and use it in other moment.
But the TC2 use is different.
After you install it in the wall, you long press on one of the switches until they blink and then open broadlink app, click on add remote, select TC2 (there is already a specific voice) and, then, click on configure. After that, the switch in the wall stop blinking and in the app appear a new section with the new command (ON/OFF) for every single switch.
Anyone knows how can I intercept the code in configuration time of TC2?
Thanks

vicfergar, 2016-12-04:
Hi alesms,

I don't think that this could be possible. I mean, RM PRO is designed to learn IR codes and expose them to the app, but works different with RF codes. RM PRO firmware probably knows what codes or commands supports a TC2. 

Anyway, once the TC2 is configured and working using the Broadlink app, you can use an UDP packet sniffer to get record the communication between the app and the RM PRO. Decode the content using the key provided by the python script and try to figure it out what the data packet means.

Hope it helps to you!
Regards

mjg59, 2016-12-05:
Without having hardware, I think the only way to figure this out is with a packet dump of the entire setup process.

alesms, 2016-12-05:
Hi all,
I solved the problem using a universal remote command (e.g. the remote for the gate or similar)!
i put the remote control in learning mode and i learn the RF code sent from the Broadlink RM to Broadlink TC2 (consider that the remote control must stay very very near the RM Pro...pratically you must touch all the surface of the RM Pro while send the command to TC2 until the remote command find the RF) after the remote command learned the signal, you can turn on light with it.
At this point you can use the remote control to teach the RM Pro for the single RF command and after, code by code you can teach all the codes of the TC2 :)

closed 2016-12-10 by alesms

NightRang3r, 2016-12-29:
HI All 

I wrote a quick and dirty script to dump all codes from the Android e-Control app db file.

https://github.com/NightRang3r/Broadlink-e-control-db-dump

You can view the full discussion that led to the creation of this script here:

https://github.com/ericmagnuson/rmcontrol/issues/4#issuecomment-269480450

bear1180, 2018-04-25:
Hi Alesms, quick question, i am looking for an universale Remote Control to fix the issue with the broadlink TC2 as well. I bought 2 different kinds of that Garage door Controller but they so not work. please can you send me a link of the Remote Controller that you had used?
Thank you in advance

alesms, 2018-04-25:
Hi,
I used a remote control i already had.
Try with this one
https://www.amazon.it/gp/product/B00WINCEBU/ref=oh_aui_detailpage_o08_s00?ie=UTF8&psc=1

bear1180, 2018-04-25:
Thanks for the quick Response. I ordered that:
https://www.amazon.de/Fernbedienung-Sender-Lichtschalter-Controller-Wireless-As-picture-shown/dp/B078LXZ815/ref=redir_mobile_desktop?_encoding=UTF8&psc=1&ref_=yo_ii_img

But it is not possible to Start the learning mode.

alesms, 2018-04-25:
I do not think the problem is a remote control that not work, but maybe
you're  wrong the way you try to enter in the learning mode.
I don't t remember how to enter but keep finding on internet

com-xuonghuynh, 2020-05-27:
> Hi all,
> I solved the problem using a universal remote command (e.g. the remote for the gate or similar)!
> i put the remote control in learning mode and i learn the RF code sent from the Broadlink RM [...]

Just confirm that it not work with my RM4 Pro and Broadlink app. My remote can learn other remote but can't learn on off signal from broadlink. e-Control app is very old and it outdate for my RM4 Pro. Any solution for this TC2 ?