mjg59/python-broadlink #23: Any plan for Rm2 RF feature

tonyzhong84, 2016-12-05:
Rm2 support RF, would be great to have it

aldodemi, 2016-12-05:
You already have it, you can learn/register RF code exaclty as an IR code

tonyzhong84, 2016-12-07:
Tried several time, but I'm not able to register RF code

tyjtyj, 2016-12-10:
RF 433 and 315 works on RM2. 

Are you able to learn IR code ?

flyingartichoke, 2016-12-12:
I'm having trouble with this as well.  I can successfully learn an IR code with my Broadlink Pro but I cannot get it to work with an RF remote.  Are there any suggestions/guides for how to do this?  

When learning new RF codes through the Broadlink app itself, it had me do an RF frequency scan followed by learning the actual code.  Is there an analog to that process here?  

Or should I be using my phone + Broadlink app to generate the codes that are being learned by the python script?

tyjtyj, 2016-12-12:
I have 3 rf item 
1x 315 
2x 433

I use this code

Create a dir/folder name 'codes'

```
def learn(myrm2):
	while True: 
		codeName = raw_input("DeviceName :")
		if codeName == "":
			break
		while True: 
			myrm2.enter_learning()
			sys.stdout.write('Listening')
			for i in range(5):
				time.sleep(1)
				sys.stdout.write('.')
			ir_packet = myrm2.check_data()
			if ir_packet != None:
				break
			print "Nothing receive, try again"
			# record learned hex code to file
		myhex = str(ir_packet).encode('hex'); 
		print ''
		print str(ir_packet).encode('hex')
		f = open('codes\\' + codeName +".txt",'w')
		f.write(myhex)
		f.close()

myrm2 = broadlink.discover(1)
myrm2 = devices[0]
myrm2.auth()

learn(myrm2)

```

it will save the codes to codes/[devicename]txt

```
hex_data = [yourhexdata from file]
myrm2.send_data(hex_data.decode('hex'))
```

technicalpickles, 2016-12-12:
As said, RF support is already built in.

However, when I setup my device originally, I used the e-control app initially. There was a firmware update that said it improved RF learning capabilities.

It is working for me. In practice, I've found it necessary to `check_data()` in a loop until a code is found.

NightRang3r, 2016-12-29:
HI All 

I wrote a quick and dirty script to dump all codes from the Android e-Control app db file.

https://github.com/NightRang3r/Broadlink-e-control-db-dump

You can view the full discussion that led to the creation of this script here:

https://github.com/ericmagnuson/rmcontrol/issues/4#issuecomment-269480450

tecteun, 2017-01-18:
I found that learning a 315Mhz code only works through the app, there are two options in the app:
- learn
- sweep / learn
The 315Mhz RF only works with the sweep/learn option.

How to trigger the sweep / learn from the api? (right now I've worked around this by copying the hex codes from the android db)

luketoh, 2017-02-20:
Apparently I have a Broadlink RM3 Pro instead of the usual RM2 Pro.  Does this python script work with the RM3 Pro?  I tried running a simple python program to scan for it, but I don't think it detects the RM3 Pro.  Can someone help?  Here's what I got...

pi@hassbian:/home/rmcontrol $ python broadlinktest.py
discover
Traceback (most recent call last):
  File "broadlinktest.py", line 8, in <module>
    devices[0].auth()
IndexError: list index out of range

here's the program i used from someone else...

#!/usr/bin/python

import broadlink
import time

print "discover"
devices = broadlink.discover(timeout=5)
devices[0].auth()
print "enter_learning"
devices[0].enter_learning()
print "sleep"
time.sleep(5)
print "check data"
ir_packet = devices[0].check_data()
print str(ir_packet).encode('hex')
devices[0].send_data(ir_packet)
print "wait"
time.sleep(2)
devices[0].send_data(ir_packet)

pavram, 2017-03-06:
Someone might need to dump the data from the app communicating with an RM3 to get the "Sweep Learning" command structure.  It sounds like RM3 pro might have a few new commands that work in a slightly different manner.

The RM3 pro should still be detected by the Discover requests however, since discover is common among all the devices.

@luketoh: it does appear that your RM device isn't being detected.
(the devices[0].auth() command is trying to get the 0'th [first] device from the discovered devices, and that device is "null" (or not found)).

In your case I would make sure the device was on the same network as the computer you are using to attempt to detect it.  

Easy way is to use the official app and associated guides to make sure it is on your home wifi, and make sure your computer is on the same network, and work from there.

lprhodes, 2017-04-19:
RM3 (and possibly RM2) works by first getting the frequency and then learning the specific code. This code only performs the learning from the look of it.

Working on getting it working on the JS version of the code (broadlinkjs) so perhaps you can get it working in python from that once I'm done - I'll keep you posted

lprhodes, 2017-04-19:
Struggling to get the RF sweap learning working @mjg59 perhaps you can see something I'm missing

Wireshark shows the following hex as being sent when I first enable the RF sweep: 
5aa5aa555aa5aa5500000000000000000000000000000000000000000000000029d200002a276a00b58128d7e734ea3401000000c8be0000065d76df227ec2b694e98c88215afff3

The command part of the hex is 65d76df227ec2b694e98c88215afff3
Deciphered that's 9fa6f2de6d22144b0b1ce7d551a8c485

When I try to reuse that the code produces the following hex:
5aa5aa555aa5aa550000000000000000000000000000000000000000000000000dd100002a276a00020028d7e734ea3402000000d7c60000065d76df227ec2b694e98c88215afff3

As far as I can see, the the only differences between the two are the id (which I assume is the ID for the device assigned to the device sending the hex codes to the RM device) the count (which is always going to be different between the two) and the checksum - which would of course be different.

The problem is, my own command fails to enable the RF sweep mode

lprhodes, 2017-04-19:
Made the tiniest bit of progress - the hex payload for starting the scan is 19 with a command of 6a

lprhodes, 2017-04-19:
Cancelling looks to be a hex payload of 1e

prodigy7, 2017-04-19:
@iprhodes: I've already ported the whole code (based on python-broadlink Dec 2016) to javascript and it is also working. I've stopped further development because of missing time and some problems which asynchron calls. Later I'll post a address where you can the current code, maybe it will help you.

lprhodes, 2017-04-19:
@prodigy7 how does it differ to broadlinkjs which is also a port? Unfortunately neither  support the RF scanning

mjg59, 2017-04-22:
I'm afraid I don't have any RF devices, so the only feedback I have is from what other people have worked out. If you make any progress I'll happily merge it!

lprhodes, 2017-04-27:
@mjg59 The first stage (sweeping part) sends "0x19" to start the RF scanning then repeatedly sends "0x1a". You can cancel it by sending "0x1e"

The response from checking the data starts with "0x26" and then you need to check 0x4 of the response payload - if it is equal to 0x1 then sweeping has done what it needs to do.

We then go to the next stage of learning the button press...which I'm still figuring out.

xSeekTruthx, 2018-03-20:
Would love for my RM Pro to learn RF without having to use android apps. Any progress?

DODMax, 2018-07-12:
Would love to get that working too, how do I filter on Wireshark?
Tried by IP and MAC address but no luck. Thanks.

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/218. Thank you!

closed 2021-02-17 by felipediel