mjg59/python-broadlink #24: Missing pycrypto dependency

technicalpickles, 2016-12-05:
I had this library working on with a script one machine, and when moving to another, I got this error:
```
Traceback (most recent call last):
  File "send.py", line 3, in <module>
    import broadlink
  File "/home/pi/picklehome/lib/python3.5/site-packages/broadlink/__init__.py", line 4, in <module>
    from Crypto.Cipher import AES
ImportError: No module named 'Crypto'
```

Found the fix as  `pip install pycrypto` on http://stackoverflow.com/questions/19623267/importerror-no-module-named-crypto-cipher . That suggests to me it just needs to be a dependency.

Danielhiversen, 2016-12-18:
@mjg59 Could you add pycrypto as a dependency?

mjg59, 2016-12-25:
Fixed

closed 2016-12-25 by mjg59

thomasf, 2017-04-19:
pycrypto is unmaintained and should not be used. https://github.com/Legrandin/pycryptodome should be used instead... On top of being unmaintained pycrypto doesn't install correctly on a bunch of platforms