mjg59/python-broadlink #27: How to convert LIRC codes to something useable by python-broadlink?

voltagex, 2016-12-11:
Given something like https://gist.github.com/voltagex/1546899, how could I convert this to something send_data would accept?

e.g. in Python 3:

```python
>>> s = '000000000038BC81'
>>> on = codecs.decode(s,'hex_codec')
>>> devices[0].send_data(on)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/broadlink/__init__.py", line 376, in send_data
    self.send_packet(0x6a, packet)
  File "/usr/local/lib/python3.5/dist-packages/broadlink/__init__.py", line 215, in send_packet
    payload = aes.encrypt(bytes(payload))
  File "/usr/lib/python3/dist-packages/Crypto/Cipher/blockalgo.py", line 244, in encrypt
    return self._cipher.encrypt(plaintext)
ValueError: Input strings must be a multiple of 16 in length
```

tyjtyj, 2016-12-11:
hey.
i did some decoding here. u might wanna read up
https://github.com/mjg59/python-broadlink/issues/10#issuecomment-263129225

bengtmartensson, 2017-06-03:
(I am the author of IrScrutinizer.)

I just implemented a Broadlink-python export format for IrScrutinizer, which should do exactly what you ask for. You can generate python code for sending your Lirc commands to the Broadlink. Let me know if it works for you.

raman325, 2018-10-25:
@bengtmartensson I just downloaded IrScrutinizer to try this, how do I configure the Export tab to generate the signal in a format I can use with Broadlink-python?

bengtmartensson, 2018-10-25:
Selecting "Broadlink Python IR" as export format?

raman325, 2018-10-25:
I don't see it as an option using Scrutinzer 1.3. Do I need a different version or to build from source?

raman325, 2018-10-28:
> The Broadcom export is supported in version 1.4.2 and later. (I *think*
> my site is clear on that the latest version are available on Github only.)
>
> Bu sure to select menu entry File -> Reset properties when you have
> installed the new version.

Just an FYI, the Mac app link points to your site and it’s the 1.3 version.
I downloaded version 1.4.2 and was able to see it. Thanks!