mjg59/python-broadlink #28: Does Broadlink devices broadcast their status?

illxi, 2016-12-11:
Hi, just wondering if broadlink devices broadcast their on/off status when there is a change? It would be useful for smarthome hub to sync their status faster, instead of query them every few seconds.

technicalpickles, 2016-12-12:
I haven't dug in super deeply to authoritatively confirm, but I would say no. The broadlink itself doesn't have concept of devices, it just can learn packets, and then send packets you know.

The only thing I can think of is something else would need to keep track of the state, but the problem is that can get out of sync like if someone hits a manual button.

pickeld, 2016-12-19:
thats can be a good lead. try to route it to yourself

maddox, 2016-12-28:
Oof what? Its sending outbound traffic?

Has anyone done any research on these things at all to see what they're doing on our network?

I believe these RM devices are made by the company that was targeted in the huge DynDNS DDOS.

webmonger, 2016-12-28:
@maddox I think that's a bit of a stretch, the vulnerability used to infect IoT devices to become part of the Mirai Bot net used open ports created on the firewalls of the networks to infect the devices.

From what I've seen and tested while using the RM2 device I see no open ports and only out bound traffic from the RM. You can check if your network has devices which have opened the ports required for Mirai at https://www.incapsula.com/mirai-scanner/

The most secure way for an IoT device to communicate with the outside world is using a secured outbound connection to a trusted end point. For more info about IoT security check out https://www.owasp.org/index.php/OWASP_Internet_of_Things_Project more specifically for manufactures https://www.owasp.org/index.php/IoT_Security_Guidance