mjg59/python-broadlink #29: Problem ready temperature.

glupusti, 2016-12-14:
```
>>> import broadlink
>>> device = broadlink.rm2()
>>> device.discover()
>>> device.host
('192.168.100.10', 80)
>>> device.auth()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "broadlink/__init__.py", line 169, in auth
    response = self.send_packet(0x65, payload)
  File "broadlink/__init__.py", line 235, in send_packet
    response = self.cs.recvfrom(1024)
socket.timeout: timed out
```

That's the error I get.
Any help?

maddox, 2016-12-17:
I'm getting this a lot too

Paxy, 2016-12-23:
Probably related issue is in case when no error was thrown but temperature with 249.0 is returned.
Is it both case of unsuccessful auth ?

felipediel, 2021-02-17:
Old issue, probably resolved. If you still have the problem, you can reopen it.

closed 2021-02-17 by felipediel