mjg59/python-broadlink #31: RM Mini 3 Blackbean

noddy119, 2016-12-19:
Over on the Domiticz forums (https://www.domoticz.com/forum/viewtopic.php?f=17&t=6929&start=100#p103517) we have been using your library to integrate broadlink.  I have a RM Mini 3 Black Bean and need to use 'broadlink.device' where the RM pro needs 'broadlink.rm'.  I supect the RM Mini 3 was released after you wrote your excellent library?

Is there a way I can supply you with the devicetype hex so you can midify the library so the same code can be used?

Secondly; can I use your code to return the mac address following a device discovery?  I worked out how to return the ip address.  This would save having to look at the router etc.

mjg59, 2016-12-20:
Yes, please just send the devtype that's passed to gendevice() and I'll add it.

noddy119, 2016-12-22:
How would I do that?

JayPearlman, 2016-12-22:
@noddy119 You can modify the code to print out the "devtype" in the middle of the discover call. You would add a new line 111 - "print devtype". Then a recompile/install of the module.

My mini 3 black bean returns 10039 or 0x2737 in hex, which is already in the code and is seemingly properly handled by the discover function. 

I think some confusion may be coming from the scripts over at domoticz.com which are skipping the discover function and creating the device in a more manual way.

Hope this helps.

noddy119, 2016-12-23:
Hi Jay - I was using an old module version - doh!  Yes my mini 3 black bean returns 10039.  With the module updated I can use 'broadlink.rm' instead of 'broadlink.device'

closed 2016-12-23 by noddy119