mjg59/python-broadlink #33: switch not functioning

claudioita, 2016-12-20:
When calling a switch, I receive the following error. Pycrypto has been installed in virtual environment with no errors

16-12-20 12:45:08 homeassistant.core: Bus:Handling <Event call_service[L]: service_data=entity_id=['switch.asus_router'], service_call_id=1977471056-8, service=turn_on, domain=switch>
16-12-20 12:45:09 homeassistant.core: Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/components/switch/broadlink.py", line 148, in _sendpacket
    self._device.send_data(packet)
  File "/home/hass/.homeassistant/deps/broadlink/__init__.py", line 376, in send_data
    self.send_packet(0x6a, packet)
  File "/home/hass/.homeassistant/deps/broadlink/__init__.py", line 235, in send_packet
    response = self.cs.recvfrom(1024)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/tasks.py", line 233, in _step
    result = coro.throw(exc)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/core.py", line 1049, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/components/switch/__init__.py", line 87, in async_handle_switch_service
    yield from switch.async_turn_on()
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/helpers/entity.py", line 353, in async_turn_on
    None, ft.partial(self.turn_on, **kwargs))
  File "/usr/lib/python3.4/asyncio/futures.py", line 388, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.4/asyncio/tasks.py", line 286, in _wakeup
    value = future.result()
  File "/usr/lib/python3.4/asyncio/futures.py", line 277, in result
    raise self._exception
  File "/usr/lib/python3.4/concurrent/futures/thread.py", line 54, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/components/switch/broadlink.py", line 134, in turn_on
    if self._sendpacket(self._command_on):
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/components/switch/broadlink.py", line 157, in _sendpacket
    return self._sendpacket(packet, max(0, retry-1))
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/components/switch/broadlink.py", line 148, in _sendpacket
    self._device.send_data(packet)
  File "/home/hass/.homeassistant/deps/broadlink/__init__.py", line 376, in send_data
    self.send_packet(0x6a, packet)
  File "/home/hass/.homeassistant/deps/broadlink/__init__.py", line 214, in send_packet
    aes = AES.new(bytes(self.key), AES.MODE_CBC, bytes(self.iv))
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/Crypto/Cipher/AES.py", line 95, in new
    return AESCipher(key, *args, **kwargs)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/Crypto/Cipher/AES.py", line 59, in __init__
    blockalgo.BlockAlgo.__init__(self, _AES, key, *args, **kwargs)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/Crypto/Cipher/blockalgo.py", line 141, in __init__
    self._cipher = factory.new(key, *args, **kwargs)
ValueError: AES key must be either 16, 24, or 32 bytes long

Danielhiversen, 2016-12-20:
For reference: https://github.com/home-assistant/home-assistant/issues/4997

I am not sure if we use the library wrong in HA or if it is a bug in the broadlink library.
But for me it seems that enter_learning sometime return an invalid packet.

@mjg59 : It would be great if you could have a look at this.

claudioita, 2016-12-20:
I don't know if this can be of any help, and I am also not sure about it, just keep in mind that:
- if I restart HASS (sudo reboot of my pi3) and I call the broadlink in front end I received the attached error.
- if , after restarting, wait for some time (few minutes), then I have no error

I am not 100% sure about this, but maybe this information can help

16-12-20 14:10:24 homeassistant.components.http: Serving /api/services/homeassistant/turn_off to 192.168.1.52 (auth: True)
16-12-20 14:10:24 homeassistant.core: Bus:Handling <Event call_service[L]: domain=homeassistant, service=turn_off, service_data=entity_id=switch.christmas_tree_lights, service_call_id=1977737264-15>
16-12-20 14:10:24 homeassistant.core: Bus:Handling <Event call_service[L]: domain=switch, service=turn_off, service_data=entity_id=['switch.christmas_tree_lights'], service_call_id=1977737264-16>
16-12-20 14:10:24 homeassistant.core: Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/tasks.py", line 233, in _step
    result = coro.throw(exc)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/core.py", line 1049, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/components/switch/__init__.py", line 91, in async_handle_switch_service
    yield from switch.async_turn_off()
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/helpers/entity.py", line 363, in async_turn_off
    None, ft.partial(self.turn_off, **kwargs))
  File "/usr/lib/python3.4/asyncio/futures.py", line 388, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.4/asyncio/tasks.py", line 286, in _wakeup
    value = future.result()
  File "/usr/lib/python3.4/asyncio/futures.py", line 277, in result
    raise self._exception
  File "/usr/lib/python3.4/concurrent/futures/thread.py", line 54, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/components/switch/broadlink.py", line 139, in turn_off
    if self._sendpacket(self._command_off):
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/homeassistant/components/switch/broadlink.py", line 148, in _sendpacket
    self._device.send_data(packet)
  File "/home/hass/.homeassistant/deps/broadlink/__init__.py", line 376, in send_data
    self.send_packet(0x6a, packet)
  File "/home/hass/.homeassistant/deps/broadlink/__init__.py", line 214, in send_packet
    aes = AES.new(bytes(self.key), AES.MODE_CBC, bytes(self.iv))
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/Crypto/Cipher/AES.py", line 95, in new
    return AESCipher(key, *args, **kwargs)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/Crypto/Cipher/AES.py", line 59, in __init__
    blockalgo.BlockAlgo.__init__(self, _AES, key, *args, **kwargs)
  File "/srv/hass/hass_venv/lib/python3.4/site-packages/Crypto/Cipher/blockalgo.py", line 141, in __init__
    self._cipher = factory.new(key, *args, **kwargs)
ValueError: AES key must be either 16, 24, or 32 bytes long

tyjtyj, 2016-12-20:
```
File "/srv/hass/hass_venv/lib/python3.4/site-packages/Crypto/Cipher/blockalgo.py", line 141, in init
self._cipher = factory.new(key, *args, **kwargs)
ValueError: AES key must be either 16, 24, or 32 bytes long
```

This line means your raw data sent is incorrect size. 
do you use hex data ?

maddox, 2016-12-21:
I tried applying the patch from #35 to see if it would help with this issue.

I tried to replicate the issue and get the `ValueError` error. What I found was that basically nothing happened when triggering switches. Nothing went on, or off. It appeared like it wasn't even attempting to contact the broadlink. Maybe I'm missing something?

Danielhiversen, 2016-12-21:
When #35 is merged, we can add the functionality to HA. But the patch will not change anything alone.

maddox, 2016-12-21:
Oh I see, so it adds some things that the HA component can use to protect from this issue?

Danielhiversen, 2016-12-21:
Yes, we can check if success to authorize the device. If not we can retry and we will not try to send any packets before we have successfully authorized.

Danielhiversen, 2016-12-28:
I think this can be closed now

closed 2016-12-28 by mjg59