mjg59/python-broadlink #38: Learning issues for some RF

Ben2303, 2016-12-23:
Hi, 
learning codes is not always working for me. I have rf shutters, the "going up" code is learned ok, but the "going down" gives different codes : sometimes a short one (most of the time), sometimes a long one. None of them are working when sent to the rm (although the "going up" works ok).

Same shutter in official e-remote is working fine, although I had to use the second learning button (which label is cropped, don't know the difference with the first one). Your lib is only offering one learning method, maybe that's the problem I encounter?

Thank you for your help!

mjg59, 2016-12-25:
I don't have any RF devices, so I'm afraid it's difficult for me to work on this :( Let me know if the new firmware helps, if not I'll take a quick look at the app and see if I can figure out what's going on. Can you let me know what the labels in the app appear to be?

NightRang3r, 2016-12-25:
@Ben2303 Hi,  Do you use the TC1 / TC2 switches how do you clone the RF codes?
Do I need some additional hardware?

Thanks!

Ben2303, 2017-01-02:
After updating firmware I can learn codes correctly for both actions!
Thank you for your fast answer, and sorry for the useless report :)

@mjg59 android application got updated too, and changing my phone language gave the full labels (instead of the previous "Apprendre u..." and "Apprendre c..." :p ) : left one is "learn single", and right one "learn combo".

@NightRang3r I'm using rm pro and brandless rf shutters bought in a cheap store. I don't really get your question though...

NightRang3r, 2017-01-02:
@Ben2303 Thanks, I figured it out :)

mjg59, 2017-01-02:
@ben2303 Thanks - I'll try to figure out what the difference between those functions is. Good to hear that updating the firmware helped, although it's still kind of annoying to have to tell people to do that!

cxcornel, 2017-01-06:
@NightRang3r or @Ben2303 can you please help me with your RF codes? I have [Funry light switches](https://www.aliexpress.com/item/FUNRY-Switch-Touch-Control-Double-Automation-Smart-Home-Touch-Panel-Wireless-Remote-Control-110v-250v/32740012810.html?spm=2114.13010608.0.0.khInYO), but I think the remote control is broken because I couldn't make them work. Also tried the in-app predefined codes for TC1 and TC2 but nothing. I'm using RM2 and so far everything else work fine. Thanks.

NightRang3r, 2017-01-06:
@cxcornel I don't know this brand, Is it 433mhz rf ?

Can you already control it via the e-control app ?

cxcornel, 2017-01-06:
@NightRang3r Yes it's 433mhz rf. No I couldn't make them work using the app because I have no remote from which I can learn the codes and the default ones for Broadlink TC1 and 2 don't work. I was thinking that I can use your RF Codes and this library to send them and trigger the learning process on the switches again.

NightRang3r, 2017-01-06:
@cxcornel 

1. You can try and buy something like this, to clone the rf signal and then use it to send the signal to the RM PRO and use the e-control app in learning mode : 

http://www.ebay.com/sch/i.html?_odkw=433mhz+rf+remote+control+clocopy&_osacat=0&_from=R40&_trksid=p2045573.m570.l1313.TR0.TRC0.H0.X433mhz+rf+remote+control+copy.TRS0&_nkw=433mhz+rf+remote+control+copy&_sacat=0

2. Maybe you can use the RM PRO, but I'm not sure if it will work, try to use this script to capture the RF data from the switch to the RM PRO, You will probably need to get the switch to pair mode of some sort by pressing the buttons for few seconds i am  not sure how it works with this brand. 

https://github.com/mjg59/python-broadlink/issues/23#issuecomment-266412604

tyjtyj, 2017-01-06:
@cxcornel, have you try find the remote in broadlink app ? 
in broadlink app, click on add remote on top right then there is search(magnifying glass) on top. type funry give me dose of result. try to use one of these and extract the data using NightRang3r script

or @NightRang3r can help you extract one of these.

NightRang3r, 2017-01-09:
@cxcornel @tyjtyj I've added another script to dump the codes **without the need for root access** for the device.

Both scripts can be found here:

https://github.com/NightRang3r/Broadlink-e-control-db-dump

cxcornel, 2017-01-09:
Thank you guys for your suggestions. I've tried almost everything, but the switches don't work. Can't make them learn new codes. The only solution I'm thinking about is to use my arduino, a 433 Mhz transmitter and a light sensor to brute force the code. I'll try the db dump script for the other devices that I have in the app. Great work!

frownbreaker, 2018-01-27:
I've got the FUNRY Light switches and I've set them up using a remote that I ordered with one switch.
Using a 433mHz receiver on an Arduino and a receiver demo I get this output

**Button A**
```
Decimal: 8989656 (24Bit) Binary: 
100010010010101111011000 Tri-State: not applicable PulseLength: 361 microseconds Protocol: 1
Raw data: 11268,1060,400,328,1124,404,1060,388,1060,1092,372,348,1104,352,1108,1072,392,340,1108,344,1112,1080,372,356,1100,1064,396,340,1104,1080,376,1084,372,1080,380,1072,384,352,1104,1068,384,1064,396,336,1116,336,1116,336,1116,

Decimal: 8989656 (24Bit) Binary: 100010010010101111011000 Tri-State: not applicable PulseLength: 361 microseconds Protocol: 1
Raw data: 11272,1056,396,328,1116,356,1096,364,1104,1072,388,336,1120,332,1124,1048,404,332,1120,340,1104,1076,388,340,1108,1080,380,344,1112,1064,388,1072,384,1072,388,1072,380,348,1108,1072,392,1048,408,324,1128,336,1108,356,1096,
```

**Button B**
```
Decimal: 8989652 (24Bit) Binary: 100010010010101111010100 Tri-State: not applicable PulseLength: 362 microseconds Protocol: 1
Raw data: 11272,1052,412,316,1132,328,1124,328,1128,1056,400,328,1128,328,1120,1076,384,332,1124,344,1100,1084,380,340,1112,1068,392,328,524,88,424,1040,420,1044,420,1016,428,312,1144,1052,392,332,1128,1060,392,336,1116,328,1128,

Decimal: 8989652 (24Bit) Binary: 100010010010101111010100 Tri-State: not applicable PulseLength: 362 microseconds Protocol: 1
Raw data: 11272,1076,376,348,1104,340,1120,340,1108,1068,392,328,1120,344,1112,1056,372,364,1092,352,1108,1068,388,344,1120,1044,388,348,1100,1080,384,1076,384,1072,388,1064,400,324,1124,1060,408,316,1132,1056,400,344,1104,352,1096,
```

Let me know if this helps..

cxcornel, 2018-01-27:
Thanks for your answer but found out that the RF receiver module was missing from the switches. After adding them, everything works fine.

frownbreaker, 2018-01-28:
No worries, glad you got it sorted - just a quick note for folk who are new to the Funry switches. The default programming grabs a code from the RF433mHz remote (I purchased the 4 button model with one of my switches) and uses this code for turning the switch on AND off. I wanted to use the switch for home automation so wanted a DIFFERENT button to be assigned to ON and OFF. This is possible: First clear the switches learned codes (4 chirps), then set the OFF code (2 chirps) and then set the ON code (3 chirps). The documentation with the switch is scant but enough to get things going. The learnt codes survive power outage/disconnection!

mand1015, 2018-04-10:
Hi!
@cxcornel @frownbreaker
I have been trying to get my rm pro to learn the RF from my funry switches without a physical remote but most of the remotes available in the app don’t seem to work. One of them randomly did, but it turned the lights on and off with the same “on” button. And I finally found one remote from a YouTube video but it’s in Thai. This remote “learned” it when I release switch after 2 or 3 chirps, and hitting the on/off button. 
Do you know if there’s a better remotes out there that can learn without a physical remote and is in English? 

Also, I’m trying to get it to work with amazons Alexa. But right now it seems like Alexa can only detect the devices in the IHC app, but not the e-control app. Any idea how I can set up my funry switches in the IHC app? Again, without a physical remote.  I have tried the basic lights but it just doesn’t work. The red light on my rm lights up, I activate the 1/2/3 chirps... and then nothing happens. After a while i get the error message that learning has failed.

felipediel, 2021-02-17:
Closing, since it was a firmware issue that has already been solved.

closed 2021-02-17 by felipediel