mjg59/python-broadlink #49: Possible to send WiFi setup information to new device?

walkention, 2017-01-05:
This library would be a fully stand-alone solution to the Broadlink application if it were possible to setup a new device on an existing WiFi network. I'm getting a new Blackbean RM3 Mini and will be sniffing packets to see how the WiFi information is sent to the device.

Not sure if this is something you have looked into already, but I have made a fork of the project to try my hand at adding this feature. Thanks for making this library!

walkention, 2017-01-09:
I did quite a bit of packet sniffing and determined how to get my RM3 Mini to get connected to my local wireless network via the devices built in AP Mode. I have added a setup() function to this library to setup a new device and updated the README.md and protocol.md with the appropriate information. 

The merge request is pending [here](https://github.com/mjg59/python-broadlink/pull/53).

I never would have figured out the checksum without your existing checksum work in the library!

pavram, 2017-03-06:
Thanks for this work walkention.

I've been working on c#'ifying the code here (and your setup code) and making a proper library that describes everything as best I can.

I found something last night that is really interesting in the payload you get as a reply to your wifi setup packet.

bytes 52-55 are the IP address on the new network that the RM device takes up. (but in reverse order)
bytes 56-61 are the MAC address of the RM device, again in reverse order.

What I find most interesting; is it reports the IP address on the new network back to the initiator on the BroadlinkProv wifi network.  So it will connect to a wifi network while simultaneously hosting its own AP (!!).

PlusPlus-ua, 2017-03-27:
**pavram** I'm working on C# version as well :) maybe we can join our hands somehow?

ptd006, 2017-11-08:
Hi @pavram, yes I believe both WiFi modes (AP+STA) is possible even with these low power chips.  The ESP8266 can do it too I think.

pavram, 2017-11-22:
@ptd006 Unfortunately I was working on a C Sharp version; which won't be much help for any kind of low power device.

@PlusPlus-ua  I'd classify my code on this project as an impressively hacky way to achieve what I wanted.

(I tried to emulate type-unsafe C style class definitions which doesn't really work in C#)  I approach many of my projects as learning for how to do things.  (In this case, it was learning how to use attributes to do something interesting - the result was not as clean as I was hoping it would be [its ugly]).

I may publish the repository when I have something useful to add.  (The library is working, I'm building an asp-net-core web API in order to facilitate using the library to do things.)

felipediel, 2021-03-12:
Feature added with https://github.com/mjg59/python-broadlink/pull/53. Thank you @walkention!

closed 2021-03-12 by felipediel

walkention, 2021-03-13:
Woo! Merged! Thanks!