mjg59/python-broadlink #50: RF Learning Issue

oscarliuyc, 2017-01-07:
I used the Home Assistant Services: broadlink.learn_command to learn codes by Broadlink RM Pro.  IR worked very well, but for RF, the code learnt were different every time it's learnt again, and the codes failed to control the RF device.  The RF device worked well using e-Control app.  Please help

Ben2303, 2017-01-16:
Have you checked #38 ?
Check your firmware version for available updates, it may help.

oscarliuyc, 2017-01-16:
Thanks Ben.  yes, I've checked that.  The firmware is v20025, which should be the latest version.

NightRang3r, 2017-01-16:
@oscarliuyc

Look at this thread:

https://community.home-assistant.io/t/broadlink-rm-pro-and-tc2-switch/7904/97

oscarliuyc, 2017-01-16:
@NightRang3r 

Thanks.  Seems it's the common issue of RF code sending issue.  Duplicating codes seems to be the workaround.  I notice that a new release of HA has been just launched.  Let me try and see if it's fixed or I will try the duplicating codes fix.

NightRang3r, 2017-01-16:
@oscarliuyc Even it the new HA release it won't work, in order to use the duplicated RF code (i had to duplicate it twice, others needed to duplicate it six times and then encode it to base64) you will need to edit the `/homeassistant/components/switch/broadlink.py`

and change:

`self.device.senddata(packet)
`

to:

`self.device.senddata(binascii.unhexlify(packet))`

oscarliuyc, 2017-01-16:
@NightRang3r Thank you so much for your advice.  I am using HA on Synology docker.  Do you know where the HA directory is?

NightRang3r, 2017-01-16:
@oscarliuyc I have no idea, But if its linux based you can do a simple search like:

`find / -name broadlink.py`

ggarza78, 2017-08-23:
I also have a problem, I set the Learning state on HA and the LED turns yellow but I receive no signal. 
Here is some of the information of my system:
```
/python-broadlink-master/cli$ python broadlink_discovery 
discover
###########################################
RM2
# broadlink_cli --type 0x2712 --host 192.168.1.175 --mac 1aacd011eaaa
Device file data (to be used with --device @filename in broadlink_cli) : 
0x2712 192.168.1.175 1aacd011eaaa
temperature = 25.9
```

```
python-broadlink-master/cli$ python broadlink_cli --type 0x2712 --host 192.168.1.175 --mac 1aacd011eaaa   --learn
Learning...
No data received...
```

I tried changing the code: 
`self.device.senddata(packet) `
to 
`self.device.senddata(binascii.unhexlify(packet))`

But still having no luck with learning my RF codes.
I am already using the RF remote that I am trying to configure with HA with IOS e-control app and I had no issues learning and activating the buttons.

I have not tried the Android option because I don't have an Android device on hand.

felipediel, 2021-02-17:
We added support for learning RF commands via Broadlink integration in Home Assistant. Closing.

closed 2021-02-17 by felipediel