mjg59/python-broadlink #51: check_power() on MP2 (detected as SP2) always returns True

Paxy, 2017-01-07:
Example code:

```
import broadlink
mac="B827EB8183B9".decode('hex')
devices = broadlink.sp2(('172.16.1.170', 80),mac)
devices.auth()
print devices.check_power()
```

Always returns True desn't matter what is actual state. 
set_power(*) is working correctly.

eddyyanto, 2017-01-08:
@Paxy I just tested my SP2 plug and I got the same issue as yours. I noticed that the return statement is missing something. Simply replace the return statement on **broadlink/__init__.py** line **355** and it should work:
from
`return bool(payload[0x4])`
with
`return bool(ord(payload[0x4]))`

vicfergar, 2017-01-08:
@eddyyanto this change will break Python 3 compatibility. You should check the type of payload[0x4] before doing the casting to ord as is done in other devices (A1 or MP1).

johado, 2017-01-14:
Hi,
I recently bought 3 EU style SP2 for $10 each from aliexpress, and had the same issue.
Aa @vicfergar pointed out it should probably be something like:
       payload = aes.decrypt(bytes(response[0x38:]))
-      return bool(payload[0x4])
+      if type(payload[0x4]) == int:
+        state = bool(payload[0x04])
+      else:
+        state = bool(ord(payload[0x04]))
+      return state

When looking at the mp1 check_power_raw() code I noticed that type check is for offset 0x04 but it actually uses payload[0x0e] - is that intentional?

I have a patch lying around to read out the name of the device as well - not sure if that is valid for anything else than SP2 devices though.

eddyyanto, 2017-01-17:
@vicfergar Ah, ok. Thanks for the info!

LordMyschkin, 2017-03-12:
when testing the actual master branch, It seems this issue is still open.

persuader, 2017-03-28:
Same issue here.

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/72. Thank you!

closed 2021-02-17 by felipediel