mjg59/python-broadlink #54: TC2 Switches Support

Oliv3rDog, 2017-01-09:
Great Work guys... 
Got all working on Home Assistant controller.

Next step should be adding TC2 wall switches integration???

thanks

NightRang3r, 2017-01-09:
@Oliv3rDog It is supported

Oliv3rDog, 2017-01-09:
Sorry, but I missed something...

How do I config it?
Just like SP2 ?
or by learning as a RF or IR remote command?

Thanks

NightRang3r, 2017-01-09:
@Oliv3rDog 

If you have TC2 light switches you may need to get a Universal cloning remote control key for car garage door or electric gate that works in 433Mhz, Use the broadlink e-control app turn on or off the TC2 switch while pointing the remote very close to the RM Pro in order to capture the command, after you are able to use the remote to turn on/off the TC2, You should use it to teach the command to whatever script you are using.

Another option is to use a script I wrote to dump the codes from the e-control app database
it requires you to have a rooted android phone:

https://github.com/NightRang3r/Broadlink-e-control-db-dump

I am working on another script to dump the codes from the android e-control app without root, I will probably publish it by tomorrow.

eddyyanto, 2017-01-17:
I tested TC2 with RM Pro last week, just like what you would do with an IR device and it worked as expected. With a slight difference of procedure of course:
1. Hook up TC2 power wiring and add it into the official app by long pressing.
2. Make sure you can on and off from the app.
3. Use the library to set RM Pro to learning mode.
4. Press the TC 2 on or off button.
5. Dump or save the learned code.

I got the consistent on/off value throughout a few tries and able to send the on/off codes succesfully.

uvandres, 2017-01-24:
Can we close this ticket? TC2 is already supported.

Oliv3rDog, 2017-01-24:
Yes.

closed 2017-01-24 by Oliv3rDog

narongdejsrn, 2017-02-20:
I just wrote a full tutorial on how to make this script work with TC2, check it out here

https://narongdej.sarnsuwan.com/blog/2017/02/19/Control-Broadlink-TC2-through-python/

dinhchinh82, 2018-01-03:
@NightRang3r 
I've followed your guide as below but it did not work. 

> Use the broadlink e-control app turn on or off the TC2 switch while pointing the remote very close to the RM Pro in order to capture the command, after you are able to use the remote to turn on/off the TC2, You should use it to teach the command to whatever script you are using.

I've installed e-control app on iOS and can turn on/off TC2 without any problem via e-control app but I stuck on capturing the command.
"Pointing the remote" means push a button on remote continuously then tap a virtual button on e-control app right?

Thank you.

Nightreaver, 2018-03-08:
Well yes... that guide from @eddyyanto doesnt work - at least then you only have one single RM pro, as the device is in "learn" mode it cannot send data. Anyways, this would include to theat to "itself"

As the TC is not "remote" pointing to RM doesnt make sense.
So either that "cloning" remote from garage or whatever, or dumping.

No solution for iOS out there I guess @NightRang3r

NightRang3r, 2018-03-08:
Unfortunately no, but you can install the app on an android emulator (ARM) and get the files from there

Nightreaver, 2018-03-08:
I'm currently trying with android studio emulator, but getting it connect to wifi to learn the TC2 commands didnt work yet

NightRang3r, 2018-03-08:
I meant configure everything on your iPhone and do a backup then install the app on an emulator and restore the backup

cevangelio, 2018-04-18:
I was able to get this to work an iOS device by sort of using @NightRang3r 's idea. I got my e-control app to learn the RF switch. I then backed up my e-control app in the cloud. I then installed an android emulator (Bluestacks), installed e-control app and restored from the backup. I installed after Broadlink RM Plugin lite, it recognized my e-control app immediately. I went to Code List in the RM Plugin and copied the code to turn off and on the RF switch. Using the code in node-red.

DMOGM, 2019-01-16:
@eddyyanto 
Please show me how to control TC2 switches with IR remote control.

com-xuonghuynh, 2020-05-27:
RM4 Pro not support for e-control. How to make TC2 work for now ?