mjg59/python-broadlink #59: [not an issue] devices = broadlink.discover(timeout=5) --->  Save "devices" in a file

alesms, 2017-01-27:
Hi,
I tried to save in a file the output of command  "broadlink.discover(timeout=5)" for a further use, but i have no luck.
I want to save the devices in a file (or simply a single device[0]), because i want to reuse it in another script where i send_data.
In this way I do not look back the device and authorize it, but I can directly retrieve his instance from the file and run the command send_data.

Thanks in advance.

johado, 2017-01-27:
I just pushed some changes in my fork, it adds a dev2devinfo() and devinfo2dev() functions.
dev2devinfo() can be used to convert a discovered device to a structure that can be saved to disk as JSON for instance. devinfo2dev() instantiates a device object based on a devinfo struct that could have been read from disk.  My branch contains merges from some other forks but I'm new to github and havent found out how to do a proper merge so github visualises where it came from.

alesms, 2017-01-28:
sorry but don't work.
I used your code but when i call
devices = broadlink.discover(timeout=5)
i got this error:
File "broadlink/__init__.py", line 117, in devicecreate
    return rm(host=host, mac=mac, name=name)
TypeError: __init__() got an unexpected keyword argument 'name'
And even if i change the code a little bit and solve this problem, when i call 
dev2devinfo(devices[0])
i got another some errors, for example, that devtype doesn't exist or name or type

radinsky, 2017-02-02:
See this one:
https://github.com/radinsky/broadlink-http-rest

johado, 2017-02-06:
I pushed a fix to the devinfo2dev function (a missing int() conversion). In have now extended my own cli application so it correctly restores the devices list from a json file and can send commands using the cached values which speeds thinks up.

alesms, 2017-02-10:
ok @johado , i'm trying to explane my steps:
i get your code at this address: https://github.com/johado/python-broadlink/tree/johado_master
then in my terminal i wrote:
import broadlink (this is ok)
and then
devices = broadlink.discover(timeout=5)
and this is the output
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "broadlink/__init__.py", line 215, in discover
    dev = gendevice(devtype, host, mac, name)
  File "broadlink/__init__.py", line 132, in gendevice
    dev = devicecreate(devtype, host, mac, name)
  File "broadlink/__init__.py", line 117, in devicecreate
    return rm(host=host, mac=mac, name=name)
TypeError: __init__() got an unexpected keyword argument 'name'

johado, 2017-02-10:
Sorry - I had missed the name argument for some of the various device classes (which I do not know if they support a name in the way sp2 devices do. I have added that now, and some packet parser stuff.

alesms, 2017-02-11:
ok, it's almost done!
i done these steps but at the end i had an error:

device = broadlink.dev2devinfo(devices[0])
with (open('/broadlinkInstance', 'w')) as output:
	json.dump(device, output)

with open('/broadlinkInstance') as input:
	dev = json.load(input)	

this works! but when i run
device = broadlink.devinfo2dev(dev)
i have an error...
this is a part of output:
dev = gendevice(devinfo['devtype'], host, hexstr2data(devinfo['mac']), devinfo['name'])
........
in __init__
    self.name = str(name)
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-3: ordinal not in range(128)
the device name in the json is something like this:
"name": "\u451s\u56dh\u1129\u20b8"

alesms, 2017-02-11:
ok, sorry, i'm solved!
in class device i replace
self.name = str(name)
whit this
self.name = name.encode('utf-8')

and all works!

closed 2017-02-14 by alesms