mjg59/python-broadlink #60: Connecting to S1

ksperr, 2017-02-02:
>>> import broadlink
>>> device = broadlink.discover(timeout=5)
>>> device.host
>>> device[0].host
('192.168.31.222', 80)
>>> device[0].get_type()
'Unknown'

How i can i get the state of the sensor?

johado, 2017-02-06:
Try printing the devtype in gendevice() and then add an elif devtyp == statement to handle the type. Try sp2 and see if check_power() function works.

ksperr, 2017-02-06:
Hmm..thanks :)

"
   elif devtype == 0x2722: # S1
    return sp2(host=host, mac=mac)
"

Response is:
### 

> SP2
> True
> Traceback (most recent call last):
>   File "C:\Users\ACV\AppData\Local\Programs\Python\Python35\probes\prop.py", line 7, in <module>
>     print (device[0].check_power())
>   File "C:\Users\ACV\AppData\Local\Programs\Python\Python35\lib\site-packages\broadlink-0.3-py3.5.egg\broadlink\__init__.py", line 356, in check_power
>     response = self.send_packet(0x6a, packet)
>   File "C:\Users\ACV\AppData\Local\Programs\Python\Python35\lib\site-packages\broadlink-0.3-py3.5.egg\broadlink\__init__.py", line 256, in send_packet
>     response = self.cs.recvfrom(1024)
> OSError: [WinError 10040] 
> 

I supose i need to change the command that it sends here "response = self.send_packet(**0x6a**, packet)" 
But have no idea how to know this command. How? Sniffing?

johado, 2017-02-06:
I guess so. You could try the other device types and see if you get any response for check_power and check_sensors calls.
I haven't done the decrypting of any of my own traces yet, so I don't have any tool or experience to share.

johado, 2017-02-07:
On my fork I just commited a simple pcapng parser that will decrypt the payload for one device that has previously been discovered by cli/johado_cli. Maybe it will help, I used it to find out the set_name command for SP2 devices.

Floyd1256, 2017-11-09:
I have tried with standard packets that are used to read temperature, A1 sensor values, and SP2 state, which all basically return 1360 bytes with 1 at position 0 and the rest are 0 (after decryption). I assume the decryption is different than the other models or the payload being sent or the command is wrong. Did anyone figure it out?

jazzina, 2017-11-20:
use 2048 packet size
like this https://github.com/jazzina/python-broadlink/blob/91bede13cfd2cfd91bcdfa8297b8e3dde122d9a6/broadlink/__init__.py#L284

felipediel, 2021-02-18:
Fixed with https://github.com/mjg59/python-broadlink/commit/b8cf8d073e409fafef50ad0fb5435cb23d016bb2. Thank you!

closed 2021-02-18 by felipediel