mjg59/python-broadlink #64: unknown device type of broadlink MP1

hamwong, 2017-02-14:
Hi, I just bought a OEM product from broadlnk MP1, from device ID show by router as "BroadLink_OEM-T1-d9-7f-22" I think the return of device type is difference and not in your database, so it shows your code return broadlink.device, I am not very good on coding but still interest to found, do you mind teach me how should I output the return packet and out to screen so I may know what packet is return and also inform you?

mob41, 2017-02-15:
You are trying know how the protocol works?

There's a protocol.md in the root of the repository.

hamwong, 2017-02-15:
sorry, I didn't make it clear, I mean by  using simple python and import your library

I simply use below code
print "Searched broadlink devicesg"
devices = broadlink.discover(timeout=5)
print "found devices"
print devices

than I got below result:
Searched broadlink devicesg
found devices
[<broadlink.rm instance at 0x766b8828>, <broadlink.device instance at 0x766b8968>]

I was wonder is there anyway that I can use python to **print** the return code? like how am I write in my test python script to show founded deices, how to show return value of "devtype"

mob41, 2017-02-15:
You are printing the "object" out. If you want something like serialization/JSON, recently there's something like this: https://github.com/mjg59/python-broadlink/issues/59 I don't recommend using the bench directly but copying the code.

Or you can use a ```for each``` in Python, in my understanding?

If you look at the code, you will have ```self.*``` in the instance. https://github.com/mjg59/python-broadlink/blob/master/broadlink/__init__.py#L134

```python
devices = broadlink.discover(timeout=5)

print 'Number of devices found: ', len(devices)
print 'Printing devices...'

for dev in devices:
    print '==============='
    print 'Device Host: ', dev.host
    print 'Device MAC: ', dev.mac
    print 'Device Type: ', dev.type

print 'End of printing devices'
```

Unfortunately I am not familiar with Python but Java. You can try the code above... There might some errors right?

vicfergar, 2017-02-16:
Hi @hamwong, to do so just add a print line with the devtype parameter at the begining of gendevice function.

johado, 2017-02-18:
If you fetch the johado clone and in the johado_master branch, cli/johado_cli will display info about the discovered devices (devtype willl be in decimal though)

hamwong, 2017-02-19:
HI, Thanks for all the response,
@johado, I have download your code, as I am really week on coding, do you mind tell me how to do the cli?

hamwong, 2017-02-19:
as I still don't know the devtype code, I tried modify else is MP1, although I got False when use devices[0].auth(),  I still able to control and check power state on MP1, if I don't use  devices[0].auth(), I am fail to control. final, how to retrieve power measurement value?

johado, 2017-02-19:
First build and install the library, then you should just run cli/johado_cli

hamwong, 2017-02-19:
do you mean like this?
[<broadlink.mp1 instance at 0x767f91e8>]

hamwong, 2017-02-20:
hi @johado 
do you mind explain more detail, how am I able to run cli/johado_cli?
my operating system is jessie on RPI3, with python 2.7

johado, 2017-02-20:
Looks like we have the same environment. I'm not at home so I can't  confirm if I have some unpushed changes.
Not sure if I made the file executable or if you should run python on it. Without args it should run discovery and try till get the power status and print that.

hamwong, 2017-02-20:
[07:23:20] pi@openHABianPi:~/python-broadlink/cli$ python broadlink_cli --type 0x2712 --host 192.168.1.1 --mac B4430DD97F22 --temp
Traceback (most recent call last):
  File "broadlink_cli", line 34, in <module>
    dev.auth()
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 175, in auth
    response = self.send_packet(0x65, payload)
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 250, in send_packet
    response = self.cs.recvfrom(1024)
socket.timeout: timed out

I just found out the code of cli
I guess host is the ip of RPI3, and I got fail...

hamwong, 2017-02-20:
I have to sleep, so sleepy

hamwong, 2017-02-21:
Hi @johado, 
I think I make the cli works, I keep having error untail I found at middle of your code, at line 88
local_ip = "192.168.1.1"

once changed the ip, it works and I got my devtype, it's **4EF7** for MP1, as I said before, It's OEM product

hamwong, 2017-02-21:
HI @johado,

I think I got another issue, once the cli generate a json file, I got below error, I think it may be due to the device name, default come from chinese

Traceback (most recent call last):
  File "johado_cli", line 37, in <module>
    dev = broadlink.devinfo2dev(devinfo)
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 82, in devinfo2dev
    dev = gendevice(devinfo['devtype'], host, hexstr2data(devinfo['mac']), devinfo['name'])
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 134, in gendevice
    dev = devicecreate(devtype, host, mac, name)
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 129, in devicecreate
    return mp1(host=host, mac=mac, name=name)
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 353, in __init__
    device.__init__(self, host, mac, name)
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 226, in __init__
    self.name = str(name)
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)

johado, 2017-02-21:
Se the closed issue #59 for a fix.
Change to self.name =name.encode('utf-8')

hamwong, 2017-02-24:
@johado do you mind share how to use your cli code?

closed 2017-03-01 by hamwong