mjg59/python-broadlink #65: Broadlink SDK

ksperr, 2017-02-18:
This is the API that broadlink is offering. You need to contact with them and they give you the KEY.
No PC support. No S1 support....

[OPEN API SDK.pdf](https://github.com/mjg59/python-broadlink/files/784959/OPEN.API.SDK.pdf)
[openSDK.zip](https://github.com/mjg59/python-broadlink/files/784961/openSDK.zip)

mob41, 2017-02-18:
Hmm... I don't see this have any related to the library?
This library already provides support to PC and does not require keys right?

ksperr, 2017-02-18:
I know, but maybe someone find it interesting.

johado, 2017-02-18:
Has anyone tried to som something with the MediaTek MT7681 SDK ( the chip used in att least some SP2 devices)?

mjg59, 2017-02-20:
I don't think there's anything relevant to this project here - we can't make use of the SDK for licensing reasons, and the SDK for the underlying chipset is below the level that we communicate with.

closed 2017-02-20 by mjg59