mjg59/python-broadlink #66: type of device installed to?

Matt8119, 2017-02-18:
i was reading the readme and it doesnt mention or i couldnt find what type of device i install this file to.  is this installed on RPI or a different platform?

THanks,

matthew

mob41, 2017-02-18:
You mean getting the device type? Or where the library is installed to?

The library can be either installed on Windows and Linux, where RPi is running on Linux (Debian).

Matt8119, 2017-02-18:
Installing it on a RPI running linux is exactly what I was looking to do. Perfect, thanks!

mob41, 2017-02-18:
Uh.... what you mean? I don't understand.

johado, 2017-02-18:
I use it in a RPI 3. Run python setup.py build and then
sudo pyhon setup.py install

mob41, 2017-02-18:
>I use it in a RPI 3. Run python setup.py build and then
sudo pyhon setup.py install

It should be `python setup.py install`. Just take a note about that 😄

johado, 2017-02-18:
Sorry about that - writing on a phone (with Swedish autocorrection  to make things even worse)

mjg59, 2017-02-20:
It should work on any system that has Python available.

closed 2017-02-20 by mjg59