mjg59/python-broadlink #68: Script cannot find RM3 Pro device?

luketoh, 2017-02-20:
Hi  @mjg59 

I saw this video and mine seems to be the RM3 Pro and not the RM2 Pro (I bought it from Banggood.com).
https://www.youtube.com/watch?v=RdJXQEQLoSo

My device has a similar sticker at the rear as shown here, and similar serial number
https://youtu.be/RdJXQEQLoSo?t=84

I tried running this to see if it can find my Broadlink RM3 Pro but it couldn't.  Any ideas why it isn't working for me?  Is the RM3 too new a product and python-broadlink cannot detect it?

#!/usr/bin/python

import broadlink
import time

print "discover"
devices = broadlink.discover(timeout=5)
print devices

Running the script produces

pi@hassbian:/home/rmcontrol $ python broadlinktest2.py
discover
[]

As shown above, it doesn't detect the Broadlink RM3 Pro.  Any help is greatly appreciated!

luketoh, 2017-02-20:
Please ignore above.

I found the problem...the broadlink rm3 pro was on a guest network, and eventhough it was issued an IP from the same range IP address and I found ping it from my desktop PC, but the Raspberry Pi the script is running on, couldn't ping it.  Very odd.  Anyway now I've made it visible to the Raspberry Pi, I can obtain the following.   

pi@hassbian:/home/rmcontrol $ python broadlinktest2.py
[<broadlink.rm instance at 0x766148c8>]

mjg59, 2017-02-20:
If it's showing up as a .rm rather than just .device then you're fine. I'll close this now, but please feel free to open another issue if things don't work - I don't think there's any difference in the protocol, but we can try to figure out what the problem is otherwise.

closed 2017-02-20 by mjg59