mjg59/python-broadlink #71: Smart plug that is BroadLink based

Amir974, 2017-02-28:
Hi!

I just got a local smart plug from a company called Smartgrade - which is based on BroadLink technology / chip - it's sold under a different name but it's managed via the regular e-control app which managed the RM2.

It seems this plug may be using a different firmware but again it should be based on the SP model...

I've yet to physically open the package - I may return it depending on the answer I get here :)

**What are the chances that the device will / will not be managed via this code?**
if I need to find out the code variation / payload signature or what ever it's called - **how difficult is it? what is the process for fitting a new model to this API?**

marcwagner, 2017-02-28:
I have two different plugs that are also controlled with the broadlink app and also off brand. exactly the same on the outside, but using a different protocol btw :) 

It you can use the app, chances are very high that this library will also work

johado, 2017-02-28:
Modigt the library to print the devtype in gendevice() and then do a discovery(). Try adding a the devtype in the big if/elif statement in gendevice.

Amir974, 2017-02-28:
Thanks!

My Smart plug (SmartGrade) is identified as SP Mini :)

closed 2017-02-28 by Amir974

Amir974, 2017-03-06:
Correction Smart grade plug (SmartGrade) is identified as SPMini2
(OEM branded SPMini2)