mjg59/python-broadlink #73: SP3 Night light

Dima73, 2017-03-01:
Night light status on/off is this possible?

mjg59, 2017-04-22:
No idea, I'm afraid. If the app can do this then probably, if not then unclear.

clach04, 2017-05-02:
The android app can do this. Any recommendations for doing packet capture/analysis or any other reverse engineering tips? Sniffing all traffic on the network isn't an option but I have a rooted phone but I'm not familiar with Android tools in this area.

I took a quick look at https://drive.google.com/file/d/0B65vYtefY0h2aE1LdWF5RG9sX00/view (can't recall where I stumbled on that) and night light is not covered (light control mentioned there appears to be covered by the a1 device).

mjg59, 2017-05-07:
I just use tcpdump from http://www.androidtcpdump.com/ - you'll need to adb shell into the device to run it as root.

clach04, 2017-05-08:
Thanks @mjg59, got it on my device and running (had some weird issue with chmod not working so I gave up and just issue `exec tcpdump` to get it running). Any flags you recommend? I've not used tcpdump before, I had a quick scan of http://www.tcpdump.org/tcpdump_man.html

is:

   tcpdump -A

enough, too much? I'd rather leverage your experience here than try random combos (I'm doing this on device without adb and using an screen keyboard).

mjg59, 2017-05-08:
tcpdump -i any -w /sdcard/tcpdump.out should do - that'll give a binary file that can be imported into Wireshark.

clach04, 2017-05-11:
Thanks, that worked! I tweaked it a little to restrict to the device in question:

    exec tcpdump -i any -w /sdcard/tcpdump.out  host  MY.SP3.IP.ADDRESS

And wireshark allows looking at the packets (although I've not worked out out to extract the packets out of there), export doesn't give the packet contents.

Sadly I've not had time to progress further :-(  I think the first thing needed to do is take some of the broadlink Python code, and run those packets through in reverse so that they can be decrypted and then analysed (unless anyone has something like this knocking around already).

I did get a response from support@ibroadlink.com and they are NOT interested in documenting the protocol :-( So reverse engineering seems the only option.

Floyd1256, 2017-11-08:
Any luck in your reverse engineering attempts my friend?

clach04, 2017-12-13:
Sorry for delay, some how missed this. No. I wrote a quick and dirty decrypt for another remote module (that also uses AES, but in a different mode) and some of that code can be reused (I think), see https://github.com/codetheweb/tuyapi/issues/5 for details.

Floyd1256, 2017-12-14:
@clach04 no worries, will try to look into it, not actually sure if it's worth it for this particular feature. Thanks for the help !

Nightreaver, 2018-03-13:
Just for your information #158

closed 2019-08-27 by Dima73