mjg59/python-broadlink #74: Info on using Broadlink's cloud air conditioners codes

matangg, 2017-03-01:
This project is great and it helped me out a lot so I decided to contribute back my findings about Broadlink's A/C cloud capabilities.

So Broadlink's e-control app asks to learn every key code for TV remotes since each key maps to one code corresponding to the key. For air conditioners this approach is impractical since the remote sends the entire A/C state to the A/C unit. So Broadlink has a nice feature where you add an A/C remote and it asks you to only push the power button. It detects your remote and then pulls the entire code set for the A/C unit from the cloud. @NightRang3r broadlink e-control db dump can help you dump codes that are directly learned by the app but it doesn't dump cloud A/C codes. So I researched this functionality and managed to decode the A/C unit codes. This could be useful for people wanting to use their A/C code set with specific settings with the Broadlink python script (e.g. set a/c mode to "cool" with a 23 degree temperature and fan set to "low").

I used a sniffer to see what happens on the network when the e-control app learns the A/C code. There's a weird non-http binary protocol going on port 80 and 14000 that goes out to Broadlink's servers (probably with a sample code from the remote)  and eventually it downloads your A/C codes from the following url:
http://cloud.broadlink.com.cn/irda_code/XXX.gz
Where XXX is the code for your A/C remote control. Mine was 135.gz. If you're using the e-control app you can find this file stored on your Android device in this path:
/Device storage/broadlink/newremote/SharedData/ConCode/XXX.gz

Each gzipped file extracts to exactly 1,441,068 bytes and they all have the same structure. The file contains a 108 bytes header that starts with a short string description for the remote type. I didn't decode the rest of the header values but most of them are similar for all the files. Afterwards there are 1501 fixed size IR code packets that are exactly 960 bytes each. You can use these codes directly with the send_data method. Of course the packets are usually shorter than 960 bytes so the packets ends with 0 bytes padding but you can extract the real size from the packet itself.

So the file structure looks like this:
`[108 bytes header starting with string description][960 bytes IR code 1][960 bytes IR code 2]...[960 IR code 1501]`

The Broadlink IR code packets themselves are described here by @tobiaswaldvogel:
https://github.com/mjg59/python-broadlink/issues/57

Now, I decoded all of the codes for my A/C unit and you can find them here:
[ac_codes.txt](https://github.com/mjg59/python-broadlink/files/812121/ac_codes.txt)

Each packet represents the entire state of the remote control that you are sending to the A/C unit. So for example my A/C unit has 15 degrees (15-29), 5 modes (cool, heat, auto, dry, fan) and 4 fan speeds (low, medium, high, auto) so to represent all the codes you need 15 * 5 * 4 = 300 codes. The first code seems to be something like a default code or something.

If we look at the codes themselves we see:
Code    1 Power: On  Temperature: 26 Mode: Auto Fan: Auto
Code    2 Power: On  Temperature: 15 Mode: Auto Fan: Auto
Code    3 Power: On  Temperature: 15 Mode: Auto Fan: Low 
Code    4 Power: On  Temperature: 15 Mode: Auto Fan: Med 
Code    5 Power: On  Temperature: 15 Mode: Auto Fan: High
Code    6 Power: On  Temperature: 16 Mode: Auto Fan: Auto
Code    7 Power: On  Temperature: 16 Mode: Auto Fan: Low 
Code    8 Power: On  Temperature: 16 Mode: Auto Fan: Med 
Code    9 Power: On  Temperature: 16 Mode: Auto Fan: High
Code   10 Power: On  Temperature: 17 Mode: Auto Fan: Auto
Code   11 Power: On  Temperature: 17 Mode: Auto Fan: Low 
Code   12 Power: On  Temperature: 17 Mode: Auto Fan: Med 
Code   13 Power: On  Temperature: 17 Mode: Auto Fan: High
...
Code   61 Power: On  Temperature: 29 Mode: Auto Fan: High
Code   62 Power: On  Temperature: 15 Mode: Cool Fan: Auto
Code   63 Power: On  Temperature: 15 Mode: Cool Fan: Low 
Code   64 Power: On  Temperature: 15 Mode: Cool Fan: Med 
Code   65 Power: On  Temperature: 15 Mode: Cool Fan: High
Code   66 Power: On  Temperature: 16 Mode: Cool Fan: Auto
Code   67 Power: On  Temperature: 16 Mode: Cool Fan: Low 
Code   68 Power: On  Temperature: 16 Mode: Cool Fan: Med 
...
You get the idea.

My A/C also has a "swing" and "sleep" feature but it is always set to OFF in the codes so I omitted them. I guess they are not represented by the e-control remote. The weird thing is that after 301 codes all the same codes are repeated only with the power setting set to OFF and this set is repeated 4 times up until the 1501 code. Perhaps all these slots are there in order to accommodate A/C remotes with more features.

Now since the file structure is so rigid it may be possible that all A/C codes are represented the same way as my A/C unit for all A/C units code files. In that case if one wants a certain code he can just get the code number from the ac_codes.txt file, multiply by 960 and subtract 108 bytes for the header to get the offset in the file for the specific code. Another possibility is that the header defines the number of modes, fan speeds, etc and then to get a specific code you need to calculate the offset using the number of possible states for each mode, temperature, etc. Since I have only one A/C unit type I can't really verify this. Hopefully this will get the ball rolling so we can crack this.

clkao, 2017-07-30:
@matangg Thanks for sharing your findings!  Did you get the ac_codes.txt mapping by comparing the learned codes from your A/C's remote? or was it from any other documentation?

I tested the --ha option from #120 along with code extracted by:

    cut -b109- | dd if=/dev/stdin of=/dev/stdout bs=960 count=1 skip=68 | base64

But that doesn't seem to be what Code 69 (Power: On  Temperature: 16 Mode: Cool Fan: High) is supposed to do.  Instead it's toggling power on/off, where Code 369 seems to be actually switching to 16/Cool/High.

matangg, 2017-07-31:
@clkao You're welcome.

I got the ac_codes.txt with the help of this project:
https://github.com/barakwei/IRelectra
(Thanks @barakwei you're awesome!) 

This project had the algorithm and specs on how to encode the remote control state for my A/C unit. I made a python decoder script for Broadlink's packet format and verified that I can correctly decode commands from my remote control received through RM Pro. Then using this script I decoded all the IR codes in the 135.gz file which is the file the e-control app recognized for my A/C unit.

As I said, I only had my own A/C unit for testing so I can't be sure each offset will encode the same result for other A/C units. Since each A/C file is exactly the same with 1501 codes my theory was that each "slot" might encode the same key combination for all A/Cs. Since my A/C unit has exactly 300 states and 1501 - 1 neatly divides by 300 it seemed logical. But admittedly even if the e-control remote needs all 300 possible states for "On" and the same 300 states for "Off" it still leaves us with 900 redundant codes at least for my A/C unit. And we actually see that the same set of codes in the "Off" state are repeated 4 times in the file.

So it's entirely possible that other A/C units have different min/max temperatures and/or other features so they need more slots to encode these states. My guess is that the header that I didn't parse contains information for the app about these values. If someone else can decode the entire file for a different A/C unit and post all the combinations we could possibly try and understand the values in the header and then compute the correct offset for each combination.

gadido30, 2017-08-14:
@matangg 

Can you please share python decoder script for Broadlink's packet format?

matangg, 2017-08-21:
@gadido30 asked for my python script that decodes the Electra A/C codes from Broadlink's packet format. I can't find the comment now (deleted perhaps?).

I'm really sorry but my script is an absolute hack job. I actually find it hard to decipher now that I look at it. But I'm posting it here because people might find it useful for decoding the Broadlink packet format although it is specialized specifically to decoding the Electra IR codes. I'm sure this can be done in a much more efficient and elegant manner but I wrote this while wrapping my mind around this whole Manchester coding thing.

The decodeElectra function can also take a live code from the A/C remote and decode it. This is the reason for all the number rounding going on since the actual remote doesn't produce exact timing. I round the time to the nearest time slice.

Again, thanks to @barakwei for his IRElectra project that allowed me to decode the packets.

[decodeElectra.txt](https://github.com/mjg59/python-broadlink/files/1240090/decodeElectra.txt)

mouth4war, 2017-11-10:
Hey @matangg I can help with verifying the logic if you can generate the codes for http://cloud.broadlink.com.cn/irda_code/600.gz .

Counts:
Temp Range: 13 (18-30 C)
Modes: 5 (Auto, Cool, Dry, Fan and Heat)
Fan Speed: 4 (Auto*, Low, Medium, High)

misterx, 2017-12-02:
I've pushed my findings. You can use my PHP implementation of Broadlink Cloud search by keyword for fetching needed remote.
https://github.com/ThePHPGuys/broadlink#draft-implementation-of-broadlink-catalog-cloud
Each remote is a simple zip with json files that contains all codes for remote.

xdream101im, 2018-01-12:
Hi
Thank you all. I was able to obtain the code for my Fujitsu AR-REA1E Remote control. 
I have noticed that once your remote is sync with the e-control app, it actually displays file name on the e-control's air con remote. On the iPhone it is on the top right corner under the "...".
You will then be able to pull it from the following, replacing the xxxx with 1008, which is the file for AR-REA1E.
http://cloud.broadlink.com.cn/irda_code/xxxx.gz

Based on the command provided by @clkao, I've converted it to base64 strings that HA understands.
`cut -b109- 1008 | dd if=/dev/stdin of=/dev/stdout bs=960 | base64 > AR-REA1E.txt`

While analyzing 1008, I've noted the structure is very similar to what @matangg has indicated. However, my file has only 1300 ish record and that there were a lot more padding after the 53 code. As a result, this was an indication of the end each mode section. My remote only has temp range between 18-30, the e-control app also only have the Operation Mode of [Auto, cool, heat, dry, fan], the fan speed are [low, mid, high, auto]. What this means it there are 13 x 5 x 4 = 250 combination. 

As also indicated by @matangg, the codes repeats, but instead of the OFF state repeated 4 times, the repeats is the first set [A] of codes repeated 2 times, and the third set [B] repeated 3 times.  When I tested the codes, the set [A] turns on the AC. the second set [B] only changes temperature and does not turn on or off the AC. I believe it is not sending the Power On command.

It was also identified the very first code [Code 1] as indicated by @matangg, is actually the command to turn off the AC unit. 

I have only made assumption on the mode order based on @matangg suggested format.

I have the attached the file for your review 
[1008-AR-REA1E analysis.csv.txt](https://github.com/mjg59/python-broadlink/files/1625880/1008-AR-REA1E.analysis.csv.txt)

I hope this helps others in getting the code.

minhlead, 2018-01-16:
@xdream101im i decoded my file with your command and get back the txt file with a lot of ir command padded with AAAA. But how did you filter those into the csv file?

xdream101im, 2018-01-16:
@minhlead each new command starts with a J and Ends with a series of AAAAA. You can paste it in a text editor and replace the last 5 A's to a new line.
I feed it into Excel to quickly did my analysis.

minhlead, 2018-01-17:
@xdream101im that's actually quite ingenious i tried that and the result look quite promising. Will report back later

olskar, 2018-01-29:
How is this coming along for you @minhlead @xdream101im ? got it to work?

xdream101im, 2018-03-06:
@olskar I've been using my original file with no issue on all three of my Fujitsu AC.

olskar, 2018-03-06:
@xdream101im would you mind uploading it somewhere? :)

xdream101im, 2018-03-06:
@olskar my bad... see https://gist.github.com/xdream101im/2bdf106e2fcae749b383bc45c2ce327c

andrezpt, 2018-03-12:
@xdream101im great job, i will test in mine.

Thanks

hhaim, 2018-04-13:
I've tested Electra A/R 135.xxx (Electra) codes  . "ON" is actualy "toggle". Is there a way to query the state (on/off) and only then toggle?

pickeld, 2018-04-20:
@hhaim did you manage to find the ON/OFF command? im experiencing the same issue with the 135 codes (Electra)

hhaim, 2018-04-20:
No. There is no such simple command for my A/C and there is no feedback.
 (One option is to set timer on/off with relative time of zero and start
it).
I’m moving to Sonoff solution with cold relay.
With this not only I would have simple on/off I would have a feedback to
the current status.
IR is good for feedback-less like TV Volume up/down not more than that.

Hanoh

smartnakh, 2018-09-02:
@hhaim  did you find a solution for the toggle ?

hhaim, 2018-09-02:
Yes, I’m using sonoff with a cold relay.

hhaim, 2018-09-02:
> What do you mean by "cold relay"

Electra A/C has a CLK input. When CLK is N/O -> N/C the A/C will start
working. N/C -> N/O will signal the A/C to stop.
You can’t change the temperature using this way but you can make sure the
A/C is off.

I’ve used SONOFF 4CH Pro relay output for that.

Thanks,
Hanoh

smartnakh, 2018-09-02:
i have a script to send IR to the electra AC 135

in the file shared above, for example :
Code   38 Power: On  Temperature: 24 Mode: Auto Fan: Auto
i have used the getBroadlinkSharedData.py to get the code
i have a script to send record,  for the same mode than the code 38, i have captured 2600b80066813f211f41211e221e421e221e221e221e1f41411f213e421e1f221e211f221e22201f1f221f1f211f211f211f1f211f22211e1f221e221f3f3f2161843e22203f211f211e421e221e1f21221e1f42401f1e423e221e211f22211e1f221e22201f211f211e221e1f211f22211e1f221e22201f213f3e216184411e223e221e211f411e221e221e2120213f401f223e411f211e221f211e2120211e221e221e211f1f21211f211e2120211f211e221e2040421e83000d05

when i send the long code, it works

when i am sending 38, it do not work

hhaim, 2018-09-02:
Per my experiments there is no such command ON or OFF for Electra.
ON/OFF == TOGGLE. For all the commands.
As it was designed for human IR
On top of this IR does not have a feedback for success. Suggest to look
into other alternatives.
there are RS232 or CLK input.

Thanks,
Hanoh

optimusred, 2018-11-10:
I am new to this and I have recently purchased the broadlink rm pro to use with IHC,

I did some testing and found that if I try to use the air-con template and "self-learn" buttons, this device will not show up in Google Home.

But if I used the aircon template and go through the setup using their "setup without remote" and by choosing one of the 50 IR libraries, the created "air con" can actually show up in Google home.

However, with that, though my aircon can now be controlled with Google Voice, it can't be set to "Econo" mode as I cannot find an IR code from the library that will power on the aircon with "Econo" mode.

So I did some more digging and found forum. Judging from the earlier shared Air Con codes, it seems that they never have any intention to put "Econo" mode as part of the IR blast right?

DanielO, 2018-12-01:
FWIW I tried this recently and used mitmproxy to sniff the traffic the ihc app was sending and found..

It gets a list of brands via https://d0f94faa04c63d9b7b0b034dcf895656rccode.ibroadlink.com/publicircode/v2/app/getbrand

It submits the IR code to their servers via
https://d0f94faa04c63d9b7b0b034dcf895656rccode.ibroadlink.com/publicircode/v2/cloudac/recognizeirdata?mtag=app
which returns..
{
    "error": 0,
    "msg": "ok",
    "respbody": {
        "downloadinfo": [
            {
                "downloadurl": "/publicircode/v2/app/getfuncfilebyfixedid?fixedid=34857223",
                "fixedId": "34857223",
                "fixkey": "8f922fef",
                "name": "Toshiba_5904"
            }
        ]
    },
    "status": 0
}
(in my case)

The resulting file at https://d0f94faa04c63d9b7b0b034dcf895656rccode.ibroadlink.com/publicircode/v2/app/getfuncfilebyfixedid?fixedid=34857223 is only 11232 bytes long and is not gzip compressed.

I haven't dug into the file itself very much yet.

I'm not sure what the deal is with the ridiculous server name (d0f94faa04c63d9b7b0b034dcf895656rccode.ibroadlink.com)

The histogram looks flat-ish so I am wondering if it is encrypted :-/

nenonano, 2019-01-07:
hi all 
my file is 1014.gz (http://cloud.broadlink.com.cn/irda_code/1014.gz) for a Mitsubishi Electric AC/heat pump

Temp Range: 13 (16-31 C)
Modes: 5 (Auto, Cool, Dry, Fan and Heat)
Fan Speed: 4 (Auto*, Low, Medium, High)
Swing mode: 6 (1,2,3,4,swing, Auto)

it seems that the file has the same structure above (108 header, 1501 960 codes) but I don't know how to identify each code.
I tried some of them via HA's broadlink send packet function and I see my AC react, so it seems that codes are good.. 
is there any (easy) way to identify them?
TIA

DanielO, 2019-01-07:
I subsequently looked at an alternate approach but there is quite a lot of A/C code reverse engineering in https://github.com/markszabo/IRremoteESP8266/tree/master/src

nzfarmer1, 2019-02-21:
Just a wee note to thanks those involved in this thread.

I used the parsed example file from 1008 for my Fujitsu Heatpump - decoding the base64 and encoding as hex and saving a bunch of command files so I could use the broadlink_mqtt tool.  Worked a treat and saved me having to write an interface - although a simple MQTT wrapper based on your work would be trivial.

Now need to see how reliable it is, and if it needs to be reset from time to time.

Lisheng2016, 2019-05-21:
> I subsequently looked at an alternate approach but there is quite a lot of A/C code reverse engineering in https://github.com/markszabo/IRremoteESP8266/tree/master/src

same findings,wonder if someone find ways decoding the response

timson, 2019-07-02:
> Per my experiments there is no such command ON or OFF for Electra. ON/OFF == TOGGLE. For all the commands. As it was designed for human IR On top of this IR does not have a feedback for success. Suggest to look into other alternatives. there are RS232 or CLK input. Thanks, Hanoh

Hi! I am also owner of Electra AC (mini merkazi) and I also want to use sonoff relay to control my ac (on/off). Can you tell me where I can find CLK input in AC. Is it in external RC block, or in internal AC block? May be you can suggest documentation, electronic circuits? Thanks.

hhaim, 2019-07-02:
Every Israeli A/C includes a CLK  input. It is a two lines. 
Once they will connect to each other it will notify the A/C to start, disconnect -> stop.  
It is a simple mechanism for "Shabbat timer".
I'm using  "sonoff 4 CH Pro" to control it see the full code  [here](https://github.com/hhaim/hass)

There is another way to better control it (set/get the temperature, power, etc ) throw RS485. I'm planning to explore it

nzfarmer1, 2019-07-02:
Agree, I'd really much prefer rs485.  For my Fujitsu, the option was for a unit from an outfit called Intesis who had done the hard work of reverse engineering the signals between head and compressor, and exposed them via Modbus.

However, the unit was not cheap, and funds were tight, so we opted for the RM Mini.

I've mounted the mini immediately in front of the unit, and used the codes from the numbered file downloaded from their site.  It appears to work quite reliably without the need to reset the Mini, so that's a positive.

Ultimately I wasn't going to get any positive confirmation from the device, so considered adding a power meter to ensure the device had started/stopped.  Unfortunately first device didn't work as promised so will find another (if you have a link to an Aliexpress Power Meter that Modbus works on, let me know)

aspitznyc, 2019-12-18:
Hi there! I recently switched from OpenHAB to HA and I'm still trying to get everthing working again ... So far I've downloaded the .gz file [http://cloud.broadlink.com.cn/irda_code/1122.gz](http://cloud.broadlink.com.cn/irda_code/1122.gz), but I have no idea how to convert that file into a TXT that I can later turn into a JSON file. Any help would be appreciated!

talondnb, 2020-06-08:
> Hi there! I recently switched from OpenHAB to HA and I'm still trying to get everthing working again ... So far I've downloaded the .gz file http://cloud.broadlink.com.cn/irda_code/1122.gz, but I [...]

You need to read this post: https://github.com/mjg59/python-broadlink/issues/74#issuecomment-357218128

It will explain how to decode the codes from the cloud file.

WLines84, 2020-07-09:
Hello guys, awesome discussion going on here. 
If someone could please assist me  I would appreciate.

I have followed https://github.com/mjg59/python-broadlink/issues/74#issuecomment-357218128 and was able to extract the codes from the 600.gz file 
However how can i correlate which code to the function it should do? I think i missed a step along the way.Files attached for reference.
[600.txt](https://github.com/mjg59/python-broadlink/files/4897421/600.txt)
[600.gz](https://github.com/mjg59/python-broadlink/files/4897423/600.gz)

talondnb, 2020-07-09:
> Hello guys, awesome discussion going on here.
> If someone could please assist me I would appreciate.
> 
> I have followed [#74 (comment)](https://github.com/mjg59/python-broadlink/issues/74#issuecomment-357218128) and was able to extract the codes from the 600.gz file
> However how can i correlate which code to the function it should do? I think i missed a step along the way.Files attached for reference.
> [600.txt](https://github.com/mjg59/python-broadlink/files/4897421/600.txt)
> [600.gz](https://github.com/mjg59/python-broadlink/files/4897423/600.gz)

It’s explained earlier in the thread.

Here’s one I prepared a while ago for 600:

https://github.com/smartHomeHub/SmartIR/blob/master/codes/climate/1060.json

raphaelmsr, 2021-05-16:
Hi ! I just saw this amazing thread; I am to the point where I managed to dump the codes thanks to decodeElectra.py since my ID was 135; now I am not sure what to do; 
I have a bunch of codes and I am willing to convert them for Homebridge usage, any ideas ?(I tried the jsonSubIr method but my files were empty somehow ... )

nzfarmer1, 2021-05-16:
> Hi ! I just saw this amazing thread; I am to the point where I managed to dump the codes thanks to decodeElectra.py since my ID was 135; now I am not sure what to do;

There is a python broadlink package you can use to send the data directly to a device and a bunch of utilities that use and complement this.  Probably worth going through the code and understanding how it's put together.

I've adapted an mqtt to broadlink script for my needs that uses this package, mapping topics to commands.  If you do it this way, you can send commands from any MQTT enabled IoT framework - or even just use mosquitto_pub from the command line

raphaelmsr, 2021-05-17:
> There is a python broadlink package you can use to send the data directly to a device and a bunch of utilities that use and complement this. Probably worth going through the code and understanding [...]

Hi ! I know this package I have used it already for debugging but I am trying to find a way to generate the base64 codes automatically from the XXX.gz packages