mjg59/python-broadlink #75: capturing RF codes help

Matt8119, 2017-03-06:
i am not sure how to capture RF codes with my RM PRO. i have no issues capturing IR codes, but no matter what i try, it wont register that there is any type of communication packets being sent to the outlet...

mob41, 2017-03-06:
Isn't it the same to pair RF code like IR code?

lprhodes, 2017-04-19:
@mob41  No - RF is a two step process

miquelferrarons, 2017-07-31:
I didn't manage to do it with the API only. To learn an RF code, you need to first send the command "Sweep frequency" to the RM Pro, but unfortunately this command hasn't been integrated in the API yet. I manage to do it by sending the command through broadlink's app first, and then learning the code using the API... It was cumbersome, but I managed to make it work :/

However, it wasn't easy at all. I needed to first open the app, turn on Sweep frequency, and then learn the code. It make take a few tries until it works...

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/218. Thank you!

closed 2021-02-17 by felipediel