mjg59/python-broadlink #76: how to recognize different SmartPlug2 same series ?

zyklop70, 2017-03-08:
Hello,
i have 2 SM2 smart plugs. Standalone both works fine. 
But both at same time are not secure at discover time the sequence is not the same all the time.
Depending on which device is discovered first, the sequence is not even the same.
Is there a way to define the device absolete with (for example mac address) in python script ?
My actual script to switch on/off looks like this:

#!/usr/bin/python
import broadlink
import time
devices = broadlink.discover(timeout=2)
plug = devices[0]
devices[0].auth()
devices[0].set_power(True)  # or (False) for turn off

Thanks

johado, 2017-03-09:
You can try out the johado_branch in the johado fork. 
cli/johado_cli can do what you want.
First modify the local_ip variabler to match your setup, and run it once without arguments which will detect your devices and save the info in a json file. 
If you havent named your devices using the app you should do that, then you use that name to referens till tour device when ändring setpower commands etc.

zyklop70, 2017-03-09:
now i installed the cli from your package, but no json file was createt  and i dont understand how to send command "power on" to a smart plug using this command for example:

python /home/pi/python-broadlink-master/cli/johado_cli -d 192.168.x.x  -c setpower=1

will produce error:
line 139 and line 108 while discover
Attribut Error: ´module object´has no attribute ´dev2devinfo´

What i am doing wrong ?!

johado, 2017-03-10:
Have you installed the library from my fork as well?

zyklop70, 2017-03-10:
yes.....installed...

zyklop70, 2017-03-12:
problem solved....
i forgot to rename ALL devices via App...so it hangs at the first devices name.....my son plugged a second smartplug without my knowing ...sorry....
Now it works like a charme !!! Respect !! Great Work !!!

closed 2017-03-12 by zyklop70