mjg59/python-broadlink #77: 50% chance reading data on sensor A1

beho1der, 2017-03-11:
equipment: 
broadlink RM pro3 - IP 192.168.100.54 mac 9C:4E:36:84:D9:40
Sensor A1 - IP 192.168.100.63 mac 34:EA:34:E4:88:1C

Python script:
`
#!/usr/bin/python
# -*- coding: utf-8 -*-

import broadlink
devices = broadlink.discover(timeout=2)
devices[0].auth()
print devices[0].check_sensors()
`
When run scripts 50% that there will be an error:
# nano temp.py
root@orangepipc:~/broadlink/code# ./temp.py
Traceback (most recent call last):
  File "./temp.py", line 8, in <module>
    print devices[0].check_sensors()
AttributeError: rm instance has no attribute 'check_sensors'

And 50% that there will be an error:
# ./temp.py
{'air_quality': 'excellent', 'light': 'normal', 'noise': 'quiet', 'temperature': 23.5, 'humidity': 41.4}

Tried to use a different timeout,not any result. 
I don't understand how use broadlink using MAC and IP.

mob41, 2017-03-11:
Do you have another Broadlink devices in your network?

I think the problem you have is:

## Why this is happening?
As you are using ```devices[0].check_sensors()```, with no casting check, if the ```devices[0]``` (object index ```0``` in array ```devices```) is not an A1 sensor, it will result an error as there is **NO** ```check_sensors()``` method in such device.

Suppose you have an A1 sensor and a RM2 in the network:

This will discover devices and return a array that is **NOT ORDERED**.
```python
dev = broadlink.discover(timeout=2)
```

So you will get a 50-50 chance in discovering devices.

## Solution

To solve this, we have two methods:

1. Check the dev_type
    
    ```python
    devs = broadlink.discover(timeout=5)
    for dev in devs:
        if (dev.get_type == 0x2741):
            dev.auth()
            dev.check_sensors();
    ```

2. Check is it the instance of ```a1``` class

    ```python
    devs = broadlink.discover(timeout=5)
    for dev in devs:
        if (isinstance(a1, dev)):
            dev.auth()
            dev.check_sensors();
    ```

beho1der, 2017-03-11:
I try you methods:
1. Nothing gives
root@orangepipc:~/broadlink/code# ./test.py 
root@orangepipc:~/broadlink/code# 
2. gives an error message
    if (isinstance(a1, dev)):
NameError: name 'a1' is not defined

mob41, 2017-03-11:
Sorry... I haven't print out the method return out... try my new edit...

beho1der, 2017-03-11:
Yes, I understood why there could be an error,but almost no examples. I modifity code and it's work
```
import broadlink
devs = broadlink.discover(timeout=5)
for dev in devs:
        if (dev.get_type() == "RM2"):
        	dev.auth()
        	print dev.check_temperature();
   	if (dev.get_type() == "A1"):
        	dev.auth()
        	print dev.check_sensors();
```
Small Problem that RM2 do not always see with the discovery, so it does not come from him.
Any idea why this is happening?

mob41, 2017-03-12:
I usually use 5 seconds timeout. And there are possibilities that the Broadlink devices cannot be discovered.

In my opinion, I would run the discover() for several times and combine (exclude existing) the array.

beho1der, 2017-03-12:
Perhaps this error is due to the fact that I will use RM3 version. I tried to directly request data on IP and MAC adress.
```
import broadlink
devices = broadlink.rm(host=("192.168.100.54",80), mac=bytearray.fromhex("9C4E3684D940"))
devices.auth()
print devices.check_temperature()
```
and get error:
```
Traceback (most recent call last):
  File "./br.py", line 4, in <module>
    devices.auth()
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 175, in auth
    response = self.send_packet(0x65, payload)
  File "/usr/local/lib/python2.7/dist-packages/broadlink-0.3-py2.7.egg/broadlink/__init__.py", line 250, in send_packet
    response = self.cs.recvfrom(1024)
socket.timeout: timed out
```

mob41, 2017-03-13:
I don't know whether RM3 have compatibility with this library currently.

As you have the socket timeout kind of thing, it might incompatible with this library right now. This library (according to the code) targets SP,A1,RM1-2 something but seems not reverse-engineed the proto of RM3

If you have solved the A1 sensor problem, please close the issue. 😄

beho1der, 2017-03-14:
Probably there is a specific not compatible with RM3

closed 2017-03-14 by beho1der