mjg59/python-broadlink #79: Communication without encryption

karepiu, 2017-03-20:
does anybody know if there is a way to remove encryption from the communication? I see that in code AES is used. Is there a way to omit this part and use device without auth or with auth but with plain values ?

mob41, 2017-03-20:
Totally impossible.

The only way to do so is to modify the Broadlink firmware.
Or use the RMBridge which is probably not the purpose of this hacked library.

johado, 2017-03-22:
If you provision the device (configure the WLAN setting) using the library, the key will be all zeroes last I tried it. (Probably  bug waiting to be fixed).
Close enough?

mjg59, 2017-04-22:
As far as I know the device won't work without encryption. What are you trying to do?

karepiu, 2017-04-26:
Wanted to rewrite it to groovy on SmarthingsHub which does not support AES yet.

mob41, 2017-04-26:
You might want to write a bridge to run with SmarthingsHub and use REST to connect it.

pavram, 2017-04-27:
@karepiu 

I haven't got any experience with groovy, or SmartThingsHub, but it sounds like it is based on the JVM, and can import Java modules (the documentation on STH page mentioned "default imported java modules" which implies to me it can import other Java modules directly).  

The following github appears to be a groovy-based implementation of the Java crypt libraries to give you access to AES and anything else you might need.

https://github.com/ataylor284/groovy-crypto-extensions

HTH

closed 2019-06-01 by Danielhiversen