mjg59/python-broadlink #80: Not working in Action ON/OFF of a Domoticz switch

huongminh, 2017-03-28:
The following comand 

/home/pi/domoticz/scripts/RM-Pro/python-broadlink/playcode.py /home/pi/domoticz/scripts/RM-Pro/python-broadlink/yamazen.fan.ON.OFF.txt

works in the SSH command line but when put in the on/off action of a Domoticz switch, it does not work. Checking the Domoticz log, it says: error executing the script command /home/pi/domoticz/scripts/RM-Pro/python-broadlink/playcode.py  returned 256

if I open the playcode.py and change: import broadlink into #import broadlink then there is no error in the domoticz log anymore. But of course the script does not work.

Is the issue in the broadlink library? Please help.

ijobain, 2017-03-29:
Try to:
sudo chmod +x /home/pi/domoticz/scripts/RM-Pro/python-broadlink/playcode.py

huongminh, 2017-03-30:
Yes I did but it did not work. I still got the same error message in Domoticz log:
2017-03-30 08:07:15.354 Executing script: /home/pi/domoticz/scripts/playcode.sh
2017-03-30 08:07:15.772 Error: Error executing script command (/home/pi/domoticz/scripts/playcode.sh). returned: 256

I find out that the error with Domoticz could lie in the __init__.py in which is called by the import broadlink in the playcode.py. I tried to put #import broadlink and just leave some print out line in the playcode.sh to test, then no error in the domoticz log any more. Of course the playcode.py then does not work as expected anymore. I also have other python scripts that work well with domoticz, just the playcode does not work. Any idea to fix would be appreciated.

DT27, 2017-03-31:
In python files, you need change every relative path to Absolute path.

huongminh, 2017-03-31:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; } Could you give more detail instruction? Sorry, I am a newbie. Does absolute path means /home/pi/.... playcode.py ? If so I did.

DT27, 2017-03-31:
Not just the path in domoticz.
In playcode.py, there is a line like `jsonFile = 'code.json'`, 
change it to `jsonFile = '/home/pi/domoticz/scripts/RM-Pro/python-broadlink/code.json'`

huongminh, 2017-03-31:
I could not find jsonFile = 'code.json', in my playcode.py. I used the code from this link https://www.domoticz.com/forum/viewtopic.php?f=17&t=6929&sid=e2538021c7101670043c02f4c7688a5d&start=100 . Here is all the playcode.py posted by Noddy119 on the Domoticz forum:

#!/usr/bin/python

import broadlink
import time
import sys

try:
    fileName = sys.argv[1]
except IndexError:
    fileName = 'null'

if fileName == 'null':
   print "Error - no file name parameter suffixed"
   sys.exit()
else:

   device = broadlink.rm(host=("<your RM3 IP address>",80), mac=bytearray.fromhex("<your RM3 mac>"))

print "Connecting to Broadlink device...."
device.auth()
time.sleep(1)
print "Connected...."
time.sleep(1)
device.host

file = open(fileName, 'r')

myhex = file.read()

device.send_data(myhex.decode('hex'))
print "Code Sent...."

huongminh, 2017-03-31:
Could not find the  jsonFile = 'code.json' in the __ init __.py under the Broadlink library directory as well.

DT27, 2017-03-31:
Sorry, Confused with the `playcode.py`.
Just tried your code and worked fine both in shell and domoticz。

playcode.py```
#!/usr/bin/python

import broadlink
import time
import sys

try:
    fileName = sys.argv[1]
except IndexError:
    fileName = 'null'

if fileName == 'null':
   print "Error - no file name parameter suffixed"
   sys.exit()
else:
   device = broadlink.rm(host=("192.168.123.230",80), mac=bytearray.fromhex("b4 43 0d c7 39 ec"))

print "Connecting to Broadlink device...."
device.auth()
time.sleep(1)
print "Connected...."
time.sleep(1)
device.host

file = open(fileName, 'r')

myhex = file.read().strip()

device.send_data(myhex.decode('hex'))
print "Code Sent...."
```

tvOff.txt```
260050000001279412131114121311131213131212131213123812381138123812371337123812371337121312131114111412131213111412121238123812381237123812381237120005210001284a12000d050000000000000000
```

command in domoticz: `script:///home/pi/domoticz/scripts/python/playcode.py /home/pi/domoticz/scripts/python/tvOff.txt`

result: ```
 2017-03-31 13:18:01.917 User: 414120 initiated a switch command (38/test/Off)
2017-03-31 13:18:01.917 (RMPro) Light/Switch (test)
2017-03-31 13:18:02.898 Executing script: /home/pi/domoticz/scripts/python/playcode.py 
```

huongminh, 2017-03-31:
Thank you for your help. In my setup, it works well in shell but not in domoticz. What is your version of Domoticz? mine is not the lastest stable available. Maybe I should try to upgrade it.

huongminh, 2017-03-31:
It seems my domoticz does not work with the broadlink library code (__ init __.py),  the playcode.py itself is not the problem since I tried the command line broadlink_cli.py but it came to the same error in Domoticz.

DT27, 2017-03-31:
V3.5877, sorry can't help you more...

hddevteam, 2017-04-09:
You may try to use the lua script to call the python to send the code. Here is an example:
https://github.com/hddevteam/domoticzscripts/blob/master/callpython_device.lua

mjg59, 2017-04-22:
Hi,

I think this sounds like a problem with your Domoticz setup or install rather than the Broadlink code itself - I'm afraid I don't run Domoticz so can't easily help you.

vschagen, 2017-05-03:
I have been using you library for a few months within Domiticz. Most likely the problem is you didnt import pycryptho into python3. Using the default 2.7 works just five. The problem should reproduce over ssh from the commandline if you use python3 playcode.py

So maybe we can change the crypto lib to a more up to date version. But that's besides the issue

felipediel, 2021-02-17:
Closing, since it is not a problem to be fixed in our code base.

closed 2021-02-17 by felipediel