mjg59/python-broadlink #81: Set timer on Smartplugs

olilam, 2017-04-07:
First of all, thank you for this API! I find it very handy. 

Do you know how to add the timer functionality to the smart plugs (at least on SP3)? Do you have any plan on doing that?

I find the timer functionality quite handy to "secure" an action. For example, I can ask a switch to be ON for 1 minute. I can extend that duration by sending the command again before expiration but at least, if Wifi is lost, the switch will not stay ON too long (can be a problem if used to control a heater for example). 

By the way, how did you reverse engineer the protocol ("decompile" the official apps or analyze network traffic)?

Thanks!

johado, 2017-04-09:
Haven't really tried the timer functionality, just the schedules, but then found out that the plugs seemed to turn on and off more or less randomly, so now I have cronjobs on my Rpi3 to switch on and off the plugs, but adding a timer could be useful to get more robustness.
On my branch I have a tool to help with parsing and decrypting pcap dumps, maybe you could use that and give it ago yourself.

mob41, 2017-04-09:
In my understanding, the timer functionality is done by **the phone Broadlink app** and nothing will happen if your phone is not connected to the internet/WiFi.

Probably the timer functionality must be done by yourself via runnning a cronjob/thread as mentioned.

olilam, 2017-04-18:
Thank for your feedback. I will verify if the timer is done by the phone or by the plug itself and I will post back the result here. Of course, the real interest for me is that the plug implements the timer...

olilam, 2017-04-20:
Indeed, I have captured the network traffic and (except if I have made a mistake) the timer functionality is implemented by the phone and not by the plug.  I will now take a look at what happen with the scheduled timer. This one must be implemented by the plug...

mjg59, 2017-04-22:
I don't have any of the smartplug devices, so I can't easily add this on my own. My priority was to have enough code that platforms like HASS would be able to do their own scheduling, but I'm happy to add support if someone else figures it out.

laviua, 2017-12-13:
i tried to investigate timers and looks like device sends packet with command "setup countdown timer at xx", etc.

Nightreaver, 2018-03-13:
As the device is registering with an AWS cloud service, it is possible that they just setup a timer in the cloud to wake the device at a specific time

closed 2024-04-21 by felipediel