mjg59/python-broadlink #82: Switch timing out

scrimpys, 2017-04-10:
I have one SP2 that just won't connect. It works flawlessly with the iPhone app from broadlink but not with the python code. I have even tried setting 30 seconds as a time out and still can't see it.

There must be a more reliable way the iPhone app is connecting.  This code is one of the main reasons for me choosing the Home Assistant platform. Support for Broadlink was critical so I am very grateful but its would be awesome to have this work flawlessly.

Is there anything that can be done ?

mob41, 2017-04-10:
Can you post out your code to discover devices? 

If the iPhone app can reach the device, the problem would be either the library, your code or the network.

At first, verify whether the SP2 is running in the same local network with your computer running your Python code. It would be a problem if the computer cannot access the network that the SP2 currently in.

In detail, such that, your network has two subnet (your WiFi router is connected to another router) And your WiFi router wasn't set to have bridge mode or any static route back to the upper network. Moreover, your computer is connected to the upper network and SP2 is connected to ththe WiFi router. They won't connect and communicate together, but e-Control has its Cloud system to communicate together.

scrimpys, 2017-04-10:
Using the python code in this repository to do the discovery. Home Assistant uses the same code to do its discovery I think. 

So to be clear.
```
$ phtyon3
>>> import broadlink
>>> devices = broadlink.discover(timeout=30)
>>> print (devices)
```

output doesn't show the switch I am looking for.

SP2 is all on the same network.  Again for completeness I have a router connected to the fibre modem - wireless is off on that device.  Have 2 routers downstairs using the same SSID and 2 upstairs using the same SSID. Every other switch seems to work fine.

All other devices seem to be in the same setup as the problem switch.

Danielhiversen, 2017-04-12:
Can you try to apply this fix: https://github.com/mjg59/python-broadlink/pull/70/files ?

scrimpys, 2017-04-12:
Thanks Daniel.  Have made the changes and will monitor it over the next couple of days.  As I mentioned in an earlier post - there must be some fundament difference between the way the iPhone code works and this codes approach. Over the last couple of days I have configured my routers and devices, through mac filtering, to ensure they attach to the best/closest router.  This has improved things dramatically but I still get the occassional disconnection.  After a failed disconnection occurs it appears that HA isn't able to reconnect with it until I do a restart.

scrimpys, 2017-04-18:
Been running with the suggested changes for the last week. Seems to have less timeouts.  I say "seems" because there are quite a few variables in my environment.  I have been working to improve the WiFi in each room so this may have contributed to seeing less timeouts.  In general however I believe this change has improved things.  Noticed that sometimes a reboot of the router would leave the switch not contactable till I restarted HA.  Not sure if this is normal.

Thanks again for your hard work.

mjg59, 2017-04-22:
I've merged Daniel's fix, so with luck this should be ok now. Let me know if it's still a problem.

closed 2017-04-22 by mjg59