mjg59/python-broadlink #83: Maintenance of this library

Danielhiversen, 2017-04-12:
@mjg59 : Are you planning to maintain this library?

You have not reviewed or merged any pull request for a long time.
If you do not have any plan to maintain it, I am planning to start using my fork for Home Assistant.

mjg59, 2017-04-13:
Yeah I'm way behind. I can review some stuff over the weekend, but I'm also happy to just make you a co-maintainer?

McJono, 2017-04-14:
@Danielhiversen it would be great if you could input it into Home Assistant, if mjg59 allows. Atm the workaround is rather annoying.

Danielhiversen, 2017-04-18:
@mjg59 
I understand your limited time.
I also have a limited time to spend on these project.
If you make me a co-maintainer I can review and test bug fixes and smaller improvements.

mjg59, 2017-04-22:
I've uploaded 0.4 to pypi with all the outstanding PRs merged

closed 2017-04-23 by Danielhiversen