mjg59/python-broadlink #84: error _init_.py

Clovis-krz, 2017-04-21:
Hello,
I'm very sad because i want to use jarvis with my broadlink sp mini with domoticz but i can't use it without this software. But when i want learn.py, the concole show me this error : 

 pi@raspberrypi:~/python-broadlink $ sudo python learn.py
Connecting to Broadlink device....
Traceback (most recent call last):
  File "learn.py", line 10, in <module>
    device.auth()
  File "/home/pi/python-broadlink/broadlink/__init__.py", line 175, in auth
    response = self.send_packet(0x65, payload)
  File "/home/pi/python-broadlink/broadlink/__init__.py", line 248, in send_packet
    self.cs.sendto(packet, self.host)
socket.gaierror: [Errno -5] No address associated with hostname

I don't know what to do, have you got a idea ?

mjg59, 2017-04-22:
Where did you get learn.py from? You need to give it the IP address of the Broadlink, I think.

Clovis-krz, 2017-04-24:
I did it 3 weeks ago but nothing change. I get the learn.py in /home/pi/python-broadlink

mjg59, 2017-04-25:
learn.py doesn't come with python-broadlink, so I'm afraid you'll need to check with whoever wrote learn.py.

closed 2017-04-25 by mjg59