mjg59/python-broadlink #87: Add support for "Sweep Frequency" RF

lprhodes, 2017-04-28:
Hi @mjg59 

I have the full details on what happens when you press "Sweep Frequency" now.

These payloads are all sent with 0x6a commands.

Start Sweep Frequency: 0x19

Check for found frequency part 1 (repeat once per second): 0x1a
Wait for 0x01 as position 0x04 response:  0x1a000000010000000000000000000000

Check for found frequency part 2 (repeat once every second): 0x1b
Wait for 0x01 as position 0x04 response:  0x1b000000010000000000000000000000

Use the standard check data (release once per second): 0x04
Wait for hex code response as you would on the normal check data

mjg59, 2017-05-07:
Just to make sure I understand, the flow is:

Send 0x6a, 0x19
Send 0x6a, 0x1a until you get an 0x01
Send 0x6a, 0x1b until you get an 0x01
Send 0x6a, 0x04 to get the response

Is there any need to call 0x6a, 0x03 (the enter_learning command that's used for IR)? Does the orange LED turn on at any point in this process?

ereg, 2017-06-18:
Hello @mjg59 ,

I can confirm that the orange LED turns on, directly after entering learning "Sweep Frequency" (0x19, I guess). And it stays on during both steps.

First step is Learning frequency (0x1a ?), a long press on button until it´s done.
Second step is Learning Code (0x1b ?). and is only a short press, like with IR code.

I hope this helps, I would really love to get this working.

lprhodes, 2017-06-19:
I ended up adding it to the JS version - it should help you work things out https://github.com/lprhodes/broadlinkjs-rm/blob/master/index.js

mjg59, 2017-06-21:
@lprhodes Afraid not - it's still not clear what the correct sequence of events is. Is my understanding in https://github.com/mjg59/python-broadlink/issues/87#issuecomment-299725918 correct?

ereg, 2017-06-29:
is it possible to try this?

mjg59, 2017-11-25:
Could someone test https://github.com/mjg59/python-broadlink/tree/rf_experiment ? I don't have any RF equipment to test it with.

iondulgheru, 2017-11-29:
@mjg59, I tested and it works on python3 (3.6.3). 
But on python 2.7 I get a timeout when I run device.check_data(). After that all the calls receive a timeout and I need to reconnect to the device.
Otherwise, on python3 works as expected. I also used the returned code (device.send_data(ir_packet)) and I could control the end device.

I used this cheap RF remote to test it:
https://www.amazon.com/Lerway-Controller-Wireless-Control-R106/dp/B00AHU2U7O

masarliev, 2017-11-30:
@mjg59 Tested with some RF remotes and I get `None` on `devices[0].check_data()`
Here is example code that I use. May be I missed something
```python
import broadlink
broadlink.setup('ssid', 'password', 3)

devices = broadlink.discover(timeout=5)
devices[0].auth()
devices[0].enter_learning()
devices[0].sweep_frequency()
print("learning")
while 1:
    found = devices[0].check_frequency()
    if found:
        print('Found')
        break

while 1:
    found = devices[0].find_rf_packet()
    if found:
        print(devices[0].check_data())
```

iondulgheru, 2017-12-01:
@masarliev , try increasing your timeout to 120.
I got this timeout error also on the master branch, but after I increased it, the device was found.
If the device was already setup, and you know the IP and mac, you don't need to run discover.
Just create a device object by using:
`dev = broadlink.gendevice(DEVICE_TYPE_HERE, ("DEVICE_IP_HERE",80), mac=bytearray.fromhex("DEVICE_MAC_HERE"))`
You can get your device type from here:
`https://github.com/mjg59/python-broadlink/blob/master/broadlink/__init__.py#L16`
It's the devtype hex.

masarliev, 2017-12-03:
I don't have problem on device discovery. I don't receive the RF code

laviua, 2017-12-13:
Something goes wrong...
i've created a loop with "sweep freq (0x19)", sleep, "check freq (0x1a)" until 0x04 == 1
then find rf packet (0x1b) returns 0x04 == 1, but get learned data (0x22, 0x23) returns error...

a-lurker, 2018-02-18:
@brentavery Get rf scan learning working in CLI tool (#87) :   I had this RF code learn method working in a Vera plugin I wrote, with an RM Pro (type 0x2787) and as of about early Feb 2018, it stopped working. I now always get (@laviua)  learned data (0x22, 0x23) returns an error: where the error is always 0xfff6. Note then when doing an IR learn, the BroadLink device also returns the 0xfff6 error on get data, until it has an IR code to return.

Vera plugin:
https://github.com/a-lurker/Vera-Plugin-BroadLink-Mk2/blob/master/Luup_device/L_BroadLink_Mk2_1.lua

So do BroadLink do OTA updates automatically or do you have to manually update its firmware (note I don't use e_Control)? Maybe the firmware has changed? Is there a command to retrieve the firmware version information?

It's worthwhile running the RF code learn method without clicking on any RF remote control button. In my case I get the command: "check freq (0x1a)";  0x04 == 0x04  when it gives up looking for a RF remote command. That is, when the LED goes out. So I interpret this as "Given up on getting an RF code"

Also using the RF code learn method without clicking on any RF remote control button, I sometimes get spurious:  check freq (0x1a)";  0x04 == 0x01  indicating an RF frequency was found. It's perhaps possible the BroadLink device is picking up misc signals from around the local environment?

It also seems strange that the protocol requires a stop RF learn method when the actual device either times out (LED goes out) with no code found or stops immediately it determines a code (LED goes out). Regardless that's maybe just how its programmed.

Jeroen-R, 2018-03-11:
I can confirm that the test code from @masarliev works perfectly to pick up the RF remote for my electric gate (which requires sweep mode). Naturally you need to use the code on the [https://github.com/mjg59/python-broadlink/tree/rf_experiment](https://github.com/mjg59/python-broadlink/tree/rf_experiment) branch.

Hooked it up to the Google assistant (My code is all Python 3), I can now open the gate using voice control when half a mile away ('Open Sesame!'). By the time I drive up it is fully opened.

Saves me from reaching out to the RF remote and wait till I am in range. Saves me a good 10 - 15 seconds a day :smiley:

Is this change expected to be merged and packaged up any time soon?

okaapi, 2018-07-11:
The code from @masarliev didn't work for me....  but I got it to work with my AC114-01B RF transmitter for my projection screen, as follows:

_(I have to say that this is the first time I've used Python, so maybe its an issue of code versions, and I also can't figure out strings and bytearrays yet :-( )._

Two points:

#1 - in check_frequency there is a check 
```
if payload[0x04] == 1:
    return True
```
which never triggers. I replaced it with 
```
bb = bytearray(payload)
if bb[4] == 1 :
    return True
```
and that works for me.

#2 - calling enter_learning() before sweep_frequency() screws things up for me (as in reboot required), and find_rf_packet() should only be called once! find_rf_packet() seems to be the equivalent of enter_learning() for IR.

So it looks like this:

```
rm_sweep_frequency() # LED will be red
check_frequency()  # call until true (see #1), 1 sec intervals, requires 5-10 RF button clicks
                   # frequency is now found, and LED turns off
find_rf_packet()   # call only once, LED will turn on (just like with IR enter_learning())
check_data()       # call until result is not empty, usually requires only 1-2 RF button clicks
                   # LED will turn off, and data can be used for send_data()
```

I can share the code if somebody would like it...

My request: I have this on my Pi, and now I'd like to trigger it with Alexa...

[experiment.txt](https://github.com/mjg59/python-broadlink/files/2186468/experiment.txt)

mjg59, 2018-07-11:
Thanks, that looks like enough detail for me to implement this.

a-lurker, 2018-07-13:
okaapi -  this looks good but I'm wondering how does one detect when no frequency is found or no button is pushed after a frequency is found. And what to do at those points.

okaapi, 2018-07-13:
@a-lurker - 

1 - when no frequency is found:

You are in a loop with check_frequency(), and you can obviously give break after so many seconds. Also, the infamous \x04 position (which goes to 1 when a frequency is found) is set to 4 after 20 seconds or so, that may be a timeout. I changed my check_frequency() to return \x04 - 0, 1, or 4.

2 - if no button is pushed after the frequency is found:
At this point you're in a loop with check_data(), and you can break after, say, 10 seconds.

In both cases, the LED is still red and to get the device out of that state, cancel_learning() to the rescue!!!
```
  def cancel_learning(self):
    packet = bytearray(16)
    packet[0] = 0x1e;
    self.send_packet(0x6a, packet)
```
(I found the 0x1e command in this code https://github.com/lprhodes/broadlinkjs-rm/blob/master/index.js thank you @lprhodes )

Here's my code...
[rf_experiment.txt](https://github.com/mjg59/python-broadlink/files/2194338/rf_experiment.txt)
[broadlink.txt](https://github.com/mjg59/python-broadlink/files/2194339/broadlink.txt)

sprilukin, 2018-11-25:
I was able to successfully read RF code using provided branch [rf-experiment](https://github.com/mjg59/python-broadlink/tree/rf_experiment)

using python 3 (did not testest on python 2) with the following code snippet (edited version of @okaapi rf_experiment.txt):

```python
import broadlink
import time

device = broadlink.rm(host=("192.168.1.100", 80), mac=bytearray.fromhex("3333333333"))
device.auth()
device.sweep_frequency()
print("sweeping...; LED should be RED")

i = 0
while True:
    f = device.check_frequency()
    if not (f == 0):
        break
    i = i + 1
    time.sleep(1)
    print("check frequency... ", i)

if f == 1:
    print("frequency found! check LED, should be off")
else:
    device.cancel_learning()
    print("frequency not found... ")
    exit()
time.sleep(1)

print("learning command")

device.find_rf_packet()

i = 0
while True:
    data = device.check_data()
    if data:
        break
    elif i > 10:
        device.cancel_learning()
        print("command not learned")
        exit()
    i = i + 1
    time.sleep(1)
    print("check command... ", i)

print("command learned")

encodedData = ''.join(format(x, '02x') for x in bytearray(data))
print(encodedData)

fname = "learn_code.txt"
text_file = open(fname, "w")
text_file.write(encodedData)
text_file.close()
```

execute RF code which has been written to `learn_code.txt`:
```python
#!/usr/bin/python
import broadlink

device = broadlink.rm(host=("192.168.1.100", 80), mac=bytearray.fromhex("3333333333"))
device.auth()

file = open("./learn_code.txt", 'r')
myhex = file.read()
print(myhex)
print(bytearray.fromhex(myhex))
device.send_data(bytearray.fromhex(myhex))
```

sprilukin, 2018-11-25:
check out this PR: [218](https://github.com/mjg59/python-broadlink/pull/218)
this is almost the same as [148](https://github.com/mjg59/python-broadlink/pull/148)
but for the latest version of the library at the moment: `v0.9`,
also fixed couple issues in cli tool.

madrose, 2018-12-04:
> ```
> I can share the code if somebody would like it...
> 
> My request: I have this on my Pi, and now I'd like to trigger it with Alexa...
> 
> [experiment.txt](https://github.com/mjg59/python-broadlink/files/2186468/experiment.txt)

Hi @okaapi 
is it possible to share the codes for this remote? I have been struggling for a while now to get working codes.
Thanks in advance

okaapi, 2018-12-04:
Hey madrose,

I can send it no problem, but you might be better off starting with @sprilukin (just one or two posts up) - I have not checked his stuff yet, but it seems to work, and I'll replace my local version with his also so we're all in sync soon :-)

If that doesn't work - or you insist! - I'll send you my files.

madrose, 2018-12-04:
> Hey madrose,
>
> I can send it no problem, but you might be better off starting with @sprilukin (just one or two posts up) - I have not checked his stuff yet, but it seems to work, and I'll replace my [...]

Actually I got @sprilukin version to work, and I got some code from my RM. Thanks, but I am wondering how can I convert those codes to something I can use in Home Assistant. I believe that is base64 code (not sure ??)

rwa, 2019-01-07:
This looks like it has the potential to solve the massive headaches I have been encountering trying to learn RF codes for my RM Pro Plus 3.  I am a little unclear on the status of this.  These fixes haven't been integrated into the code yet?  Is there a low pain way to try out the feature?

sprilukin, 2019-01-07:
@rwa, in order to try out you can check out my test branch (steps are for linux and pip):
- make sure python3 is installed
- setup necessary python packages
```sh
$ sudo apt-get install python-setuptools python-dev
```
- install `pycrypto`: `$ sudo pip install pycrypto`
- checkout git repo:
```sh
git clone git@github.com:sprilukin/python-broadlink.git
cd python-broadlink
git checkout -b rf_experiment_v0.9 origin/rf_experiment_v0.9
```
- prepare dist from the source files use python3:
```sh
cd python-broadlink
python3 setup.py sdist
sudo pip3 install .
```
- to learn RF code to the file:
`python-broadlink/cli/broadlink_cli --device @<path-to-file-with-device-settings> --rfscanlearn --learnfile <path-to-file/filename>`
- to use learned file:
`python-broadlink/cli/broadlink_cli --device @<path-to-file-with-device-settings> --send @<path-to-file/filename>`

rwa, 2019-01-09:
It worked! Not well, I had to learn the code 6 times and got a different one every single time, but one of them did work.

This really should get into a release asap.

gavindsilver, 2019-04-23:
hi @sprilukin ;

I tried the above method and I am receiving this: 

Learning...
No data received...
Traceback (most recent call last):
  File "./broadlink_cli", line 186, in <module>
    dev.sweep_frequency()
AttributeError: 'rm' object has no attribute 'sweep_frequency'

sprilukin, 2019-04-23:
I'm sorry, not sure how I can help.
I just needed that stuff to work, so I digged into the code and was in the context 4 months ago.
Now I completely forgot how it works )
I do not have that environment set up to check.

closed 2019-05-18 by mjg59

steffenu, 2020-10-10:
im getting a broadlink.exceptions.ReadError: Read error

when running the script from @sprilukin 

using a broadlink pro plus_300

commit 01928b09 (2019-05-18): Merge pull request #218 from sprilukin/rf_experiment_v0.9

RF support to v0.9 resolves #87