mjg59/python-broadlink #90: Remotecentral.com

samsonrosen, 2017-04-29:
Hi!

There are a bunch of 'discrete IR codes' littering the internet specifically this page:

http://www.remotecentral.com/cgi-bin/mboard/rc-discrete/thread.cgi?5780

Is there a code converter I could use to translate it into a format the broad link device understands?

Oliv3rDog, 2017-04-29:
None that I know...
You could try using Sure Android App.
It can send a IR code from its full database through a RMPRO device and you could use another one to learn it.. or simple use an android phone with IR blaster...

mob41, 2017-04-29:
If you are interested or experienced in programming, there is a issue talking about the analysis of Broadlink's IR/RF code https://github.com/mjg59/python-broadlink/issues/57 .

You can try to write a converter for it

bengtmartensson, 2017-06-03:
You can import these codes in [IrScrutinizer](https://github.com/bengtmartensson/harctoolboxbundle). Using the export format "Broadlink-Python IR", you can generate python code for those commands. Just as an example, I impored the pages you linked to, and generated the export, [found here](https://gist.github.com/bengtmartensson/ec0af6da64f2ab8f479ab2194d9a21b0). Let me know how/if it works.

samsonrosen, 2017-06-11:
Thank you @bengtmartensson for being so helpful! I can't run this program myself. The mac app does not seem to load.

Are you able to help me again by doing this page? I think I pasted the wrong page before...

http://www.remotecentral.com/cgi-bin/mboard/rc-discrete/thread.cgi?6824

samsonrosen, 2017-06-11:
@bengtmartensson - I've managed to install it on another Mac, imported the Broadlink-Python IR. 

When I export the codes found on the page I linked and run it through BlackBeanControl I get the error "Input strings must be a multiple of 16 in length". I tried it with both your commands and mine. Do you think the original HEX could be wrong, or is it he conversion?

bengtmartensson, 2017-06-11:
> The mac app does not seem to load.

Seems to work now. [Instructions](http://harctoolbox.org/IrScrutinizer.html#Mac+OS+X+app). Possibly JDK not installed?

> When I export the codes found on the page I linked and run it through BlackBeanControl I get the error "Input strings must be a multiple of 16 in length".

Hmm, it works for me. How exactly do you invoke it? What is "BlackBeanControl"?

> Do you think the original HEX could be wrong

No, IrScrutinizer barfs at  invalid signals.

samsonrosen, 2017-06-11:
Hi,

BlackBeanControl is a python script that lets you capture IR codes and then trigger them as a saved command.

I am pasting the exported results from IrScrutinizer into BlackBeanControl where I am receiving that error. An example of something BlackBeanControl will capture looks like this:

`2600580000013094141314121512141215121413141314121436143614371436143614361536143614361536143614131412141314361413141214131412143714361436141314361400051900012f4915000c670001304914000d05` compared to what I am pasting from IrScrutinizer:

`89891134113411341111111111111111111111341134113411111111111111111111111111111111113411341111111111341134113411341111111111341134111111000589`.

I'm not sure if that helps you at all?

bengtmartensson, 2017-06-11:
No, that is not how it is supposed to be used. The exporter generates a python script that is to be executed by itself. It does not use BlackBeanControl. You only need to have broadlink in your PYTHONPATH. and use the command line parameters like broadlink_cli. For example, using my original gist,

    python broadlink_python_ir_2017-06-03_21-37-37.py --host broadlink  --mac 294cf4***  'Device Code: 7.7 Function: 251'

where the last argument is the name of the command. (The silly name comes from the import.)

samsonrosen, 2017-06-11:
Oh. I misunderstood. 

Ultimately I am using the Broadlink RM with this [homebridge plugin.](https://github.com/lprhodes/homebridge-broadlink-rm)

The intention is to paste the hex-code that the RM Mini Captures from an IR signal. However I am looking to take discrete IR Codes for the internet and paste those in.

Could IR scrutiniser be used for this purpose?

bengtmartensson, 2017-06-12:
> Ultimately I am using the Broadlink RM 

... but you asked here...

> Could IR scrutiniser be used for this purpose?

Most likely yes. You only need to write, or have someone to write, an export format for that thing, along [these lines](http://harctoolbox.org/IrScrutinizer.html#Adding+new+export+formats). It is an xml file, reasonable easy to write.

mob41, 2017-06-12:
>... but you asked here...

Hmm... what is the problem?

>Most likely yes. You only need to write, or have someone to write, an export format for that thing, along these lines. It is an xml file, reasonable easy to write.

The program turns out as useless? I am wondering why the final output cannot be exported as a hex but a script?

pavram, 2017-07-30:
good news samsonren, issue 57 solves your problem.

The codes you have that are listed on that URL look like they are in "Pronto" format.

The program ([Python Gist](https://gist.github.com/appden/42d5272bf128125b019c45bc2ed3311f)) provided by appden converts from Pronto format; into the byte-string format you are looking for.  It also includes the zero-padding that makes it a multiple of 16.

I was even capable of running that python gist in an online python interpreter.

With some slight modifications you should see what I was doing here:
https://repl.it/Josj/0

(Just follow the array format at the top, you can ignore the comments (the # lines) and copy and past the commands you want in.

Click the "Run" button at the top; and it shoudl output the broadlink codes (with poadding) on the right hand side.

samsonrosen, 2017-07-30:
Wow @pavram – that's amazing and so simple to use.

Thank you so much!

pavram, 2017-07-31:
@samsonrosen Just connecting the dots that others created! Glad I could help.

pavram, 2018-03-18:
@rschuiling 
I think repl.it must have changed something.  and must have removed comments or something.

I'll make a new version.

https://repl.it/@Pavram/HastyJealousZebrafish

hopefully this one sticks around.  I'm still not sure why the previous one went away.

at the top of the code you will see an array "CodesToConvert"

in it, it has 1 big string "0000 0011 ... 0000" (it continues over many lines)
You can replace that string with the code you wish to convert.

Then click play at the top.

rschuiling, 2018-03-19:
@pavram Thanks Peter, now I understand what the script was meant to do. Now just to get the Broadlink codes working in Home Assistant's YAML... that's still a challenge it seems :)

jgcbaiz, 2018-07-18:
Hi. Not to threadjack, but could someone please point me to a tutorial or software to download remote codes to my new Broadlink RM3 mini? I can't manually teach it discrete codes for Power ON, Power OFF, HDMI2, etc, just the toggle switches on the original remote. Is this even possible? Thanks

rschuiling, 2018-07-18:
@jgcbaiz My first recommendation would be the [SURE Universal Remote app](https://play.google.com/store/apps/details?id=com.tekoia.sure.activities&hl=de). Should also be available for iOS.

jgcbaiz, 2018-07-18:
@rschuiling Thank you for your fast reply! However my final purpose is to create a "macro" for certain actions and have it run by asking Alexa, so I am pretty much stuck with Broadlink's IHC. Can this software be programmed to input discrete codes? Or is there a way of controlling SURE through an Amazon Echo? Thanks again

jgcbaiz, 2018-07-19:
@rschuiling you pointed me in the right direction. I used my old Galaxy S6 (with a built in IR blaster) and donwloaded SURE. Then I used it to manually program Broadlink's RM3 using their IHC App. Thanks again!

rschuiling, 2018-07-20:
@jgcbaiz Great!