mjg59/python-broadlink #93: Device discovery returns 0 devices

neptunix, 2017-05-05:
Hi. Don't like to ask dumb questions, sorry

I've got RM3 Mini connected to wifi network. I can ping it and can successfully manage it from my phone, but

`broadlink.discover(timeout=30)` always returns no devices. Tried from linux and mac on the same network.
Is there something that I miss?

P.S. How does it get the password after initial setup from the phone if it's not connected to wifi? Bluetooth?

mob41, 2017-05-06:
Try taking a look of this issue #82, and #77

mob41, 2017-05-06:
RM3 seems to have compatibility issues with this library currently with heavy socket timeouts.

mjg59, 2017-05-07:
I'm afraid I don't have an RM3 so it's not easy for me to figure out what's going on here. Are you able to get a tcpdump from a Linux system when attempting discovery? It's possible that the discover packets are malformed in a way that works with older devices but is broken with the RM3.

neptunix, 2017-05-07:
Sure. I'll post back later. Thanks

vschagen, 2017-05-13:
I did NOT update the code since december, but for me discovering the RM3-Mini works just fine. Little detail (for me just cosmetic) its recognised as RM2, but IP and MAC are correct. I didn't go over the code, so did you change something in the discovery?
I'm running the code on an RPi2 (Rasbian) using both Python 2.7 and 3.x

neptunix, 2017-05-14:
Can't figure it out, but:

I got time to dump the response, and it started to come today. I didn't change anything at all (well, except apt-get upgrade few days ago, was there something related to python (haven't noticed))

Anyway, here is the response (successful):

```
14:57:23.475693 IP 10.3.1.10.http > ubox.local.35900: UDP, length 128
	0x0000:  4500 009c 5d8b 0000 ff11 47b5 0a03 010a
	0x0010:  0a03 0101 0050 8c3c 0088 1642 0000 0000
	0x0020:  0000 0000 0300 0000 e107 390e 1107 0e05
	0x0030:  0000 0000 0a03 0101 3c8c 0000 0acf 0000
	0x0040:  0000 0700 0000 0000 0000 0000 27d2 c60d
	0x0050:  3727 0a01 030a 6eb1 e034 ea34 e699 bae8
	0x0060:  83bd e981 a5e6 8ea7 0000 0000 0000 0000
	0x0070:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0080:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0090:  0000 0000 0000 0000 0000 0200
```

neptunix, 2017-05-17:
Can't reproduce the problem anymore.

closed 2017-05-17 by neptunix

mjg59, 2017-05-17:
Thanks for the update - I don't think I've changed anything relevant, but it's good that it's working.