mjg59/python-broadlink #94: Alarm Kit decrypt error

StarMX, 2017-05-06:
type:S1 Alarm Kit
host:('10.1.1.143', 80)
MAC:27dff10d43b4
ID:03000000
`5aa5aa555aa5aa55000000000000000000000000000000000000000000000000496c00002a27ee0364a527dff10d43b403000000b0be0000be45bb74d93223a22ee69226025aa5ae3d8ac1ad90952cd59f597485912
8912d20c380d8f18595b6454639de8e5bd172af7e1b9ab3b5cec2ccbfefa7c98fda9956a0dfc290c7a33fc7173cdf4b63813566bd4882734552dfaae5d6ee5dd59392bea286bdcfd581f027fbe0815314bc48592272
e27065595db6f33a6b3389b958db037028c6fe4ea1529b56c194124e2a6149b28ecd493a0e8a6fc70bddf886e9dd9ce2a8fa21e96eb2dad406dd0d6b2c813d9e5b2ba3e4d1eb213c7d2be4da35ca72cde8e1882862c
45942bdbc044f2f1903a54506cd2f218d3f66e2d804894cc1cba999247c2b12636fb4ffa10f688d885db3c57553004de4d3a5e59ab378673a1fc65eee8ba4a41195a4159f657f045211e27b372021a8940b569793e6
2b784d96637a82b537eda08176a45d9ca21057a91988bfd2b34c6f540cb874bd4812030db13ba1b2de88ba80c8bec1fb03979b59d31eb417e44b6afb4d8dbd12a9aac16be00b982e295f53d61d317bafa22534b38f1
a4e47abc618881d8ab946480f3a201f7d30ce13ddd3c84aaa95b10e88792d15221cd8cff44da5b1251d871d0a18be49f231796d74f35c3da29b1e3a634b9fcdc34d5d1c33f0e4bae89f0b8605631e5fc61f90caf8b3
1d15d587bdc148a2b638f41e2e843f2eec08820ef23d509ee5bc059952158328f714e4c44c4f6cbeedc0f8f699112ac271f189c7afc06ecd4dba46038323ac8411dce775ef1c827b1522b1962206c11aaa5e0b76823
83aa6488c187dc5330f6ef2e4439cddebe915b147f3d80de0510f7ce21d48884eb88687e55f212c04f961ef2d6395a7d0bb479d2ca78a9e86934e9e545fc1efdd44aa2dd7e130279f01c20d9a1d72cb2b44208132a2
c7053865ed7cc10f07e4a03cf90721e545fa938e3f1a7fe8c9449f19122256865ff6242fe302d405097f66dd4cd85dd14b809c22a3e9031116d6f95d25f871fc2191e720b259f92b26093569fdd4845e99ec491a43d
72361602d9a24e3d245b515e9a9fdb3dbe0903faabf05a3eca05c72e7eff252f1d31b63bec764fadb8f85da4f139fdc6966f5e98e95de5afd821f631d58756528f06ff28b9b95ff743d114d0326982f09c54f7c8fd2
2fe760b059a0a41d9c18a4f478e1725dbb5d16d0e01f96d48179228c79dd4a1873dfc0f9ba159febfe545f289ea3f32de11111528b826e93fbd0ca29ade43f102a768d8b4bf1628bf5ec47a5633030a3e8e0e71d7eb
2044179787c3b05c25a2dadef0607037e258ba6900ce5028762e43f53b53679f32ddfdf2f842bfe7874d9473322f702d1a391a06afa671ad18541b5e51378e4f0f1dfe20d4e41448936d5fa1b44b34b8a8e55b0f806
00a7fbfde552276eac673b7807e3579833b228628c0659478c5186ca9ce2c93dbb4b6b35db14cae57c18a9f541916d4da26184a2269c7621485f4fadb4ee0bb548948f142057d30efcf2dc4ac26b3875bfffb7d3f0f
458432012a4b0fd8b45a3dac3c353b790cb841b1c0d0000a924f7f5b578888ed532ca85b86b01dcb4959bdbcb949f1acfbcf39e6b1ae86fd914fe13f61378c573da853dccf18de7c0076b93215f32152acb97e35290
94718d19846a49236a28c12c2a82af0ac8b1dede3b00eb83f32a2c58dfccffd46b72a7d675027ba3513a48d3529dea6a45778c243b8b203bd41b94547159c6295bce52881143ee901ee294fba1d661281f74c42dfa1
a6b587f2a83e4c998c740250ae916654ef5e15f5f329939fb9acacbf1c24c17073c7e40269fbc19e5ace84411a2c8ea6f`

StarMX, 2017-05-06:
devtype == 0x2722: # Alarm Kit S1

StarMX, 2017-05-06:
auth
`5aa5aa555aa5aa55000000000000000000000000000000000000000000000000efd900002a27e903d20427dff10d43b400000000bec700003278740e24ad027350bcb1243e86b5a8f1817654c6972fe8a498e1
673bfdfad7`
type:S1
host:('10.1.1.143', 80)
MAC:27dff10d43b4
ID:03000000
key:fa61c523fa61c523fa61c523fa61c523
iv:562e17996d093d28ddb3ba695a2e6f58
`5aa5aa555aa5aa55000000000000000000000000000000000000000000000000176c00002a27ee03d30427dff10d43b403000000b0be0000be45bb74d93223a22ee69226025aa5ae3d8ac1ad90952cd59f597485912
8912d20c380d8f18595b6454639de8e5bd172af7e1b9ab3b5cec2ccbfefa7c98fda9956a0dfc290c7a33fc7173cdf4b63813566bd4882734552dfaae5d6ee5dd59392bea286bdcfd581f027fbe0815314bc48592272
e27065595db6f33a6b3389b958db037028c6fe4ea1529b56c194124e2a6149b28ecd493a0e8a6fc70bddf886e9dd9ce2a8fa21e96eb2dad406dd0d6b2c813d9e5b2ba3e4d1eb213c7d2be4da35ca72cde8e1882862c
45942bdbc044f2f1903a54506cd2f218d3f66e2d804894cc1cba999247c2b12636fb4ffa10f688d885db3c57553004de4d3a5e59ab378673a1fc65eee8ba4a41195a4159f657f045211e27b372021a8940b569793e6
2b784d96637a82b537eda08176a45d9ca21057a91988bfd2b34c6f540cb874bd4812030db13ba1b2de88ba80c8bec1fb03979b59d31eb417e44b6afb4d8dbd12a9aac16be00b982e295f53d61d317bafa22534b38f1
a4e47abc618881d8ab946480f3a201f7d30ce13ddd3c84aaa95b10e88792d15221cd8cff44da5b1251d871d0a18be49f231796d74f35c3da29b1e3a634b9fcdc34d5d1c33f0e4bae89f0b8605631e5fc61f90caf8b3
1d15d587bdc148a2b638f41e2e843f2eec08820ef23d509ee5bc059952158328f714e4c44c4f6cbeedc0f8f699112ac271f189c7afc06ecd4dba46038323ac8411dce775ef1c827b1522b1962206c11aaa5e0b76823
83aa6488c187dc5330f6ef2e4439cddebe915b147f3d80de0510f7ce21d48884eb88687e55f212c04f961ef2d6395a7d0bb479d2ca78a9e86934e9e545fc1efdd44aa2dd7e130279f01c20d9a1d72cb2b44208132a2
c7053865ed7cc10f07e4a03cf90721e545fa938e3f1a7fe8c9449f19122256865ff6242fe302d405097f66dd4cd85dd14b809c22a3e9031116d6f95d25f871fc2191e720b259f92b26093569fdd4845e99ec491a43d
72361602d9a24e3d245b515e9a9fdb3dbe0903faabf05a3eca05c72e7eff252f1d31b63bec764fadb8f85da4f139fdc6966f5e98e95de5afd821f631d58756528f06ff28b9b95ff743d114d0326982f09c54f7c8fd2
2fe760b059a0a41d9c18a4f478e1725dbb5d16d0e01f96d48179228c79dd4a1873dfc0f9ba159febfe545f289ea3f32de11111528b826e93fbd0ca29ade43f102a768d8b4bf1628bf5ec47a5633030a3e8e0e71d7eb
2044179787c3b05c25a2dadef0607037e258ba6900ce5028762e43f53b53679f32ddfdf2f842bfe7874d9473322f702d1a391a06afa671ad18541b5e51378e4f0f1dfe20d4e41448936d5fa1b44b34b8a8e55b0f806
00a7fbfde552276eac673b7807e3579833b228628c0659478c5186ca9ce2c93dbb4b6b35db14cae57c18a9f541916d4da26184a2269c7621485f4fadb4ee0bb548948f142057d30efcf2dc4ac26b3875bfffb7d3f0f
458432012a4b0fd8b45a3dac3c353b790cb841b1c0d0000a924f7f5b578888ed532ca85b86b01dcb4959bdbcb949f1acfbcf39e6b1ae86fd914fe13f61378c573da853dccf18de7c0076b93215f32152acb97e35290
94718d19846a49236a28c12c2a82af0ac8b1dede3b00eb83f32a2c58dfccffd46b72a7d675027ba3513a48d3529dea6a45778c243b8b203bd41b94547159c6295bce52881143ee901ee294fba1d661281f74c42dfa1
a6b587f2a83e4c998c740250ae916654ef5e15f5f329939fb9acacbf1c24c17073c7e40269fbc19e5ace84411a2c8ea6fa9230fc891a913d9a26dbfe53f6c62ad26e74eeb7251aa504d9fad3669bcff674d7f92c8dd
a0e523c21cc991a4b089123e3084937aab6347478220a23b05c01a222465504586d644199c499d7b6ab8d8080f8d00ba`

pavram, 2017-05-24:
(Check out Issue #18 for more detail)

I processed your last packet with my online inspector. (had to get rid of the spaces)
http://onlinebroadlinkinspector.azurewebsites.net/Inspect/

Packet - Data_Reply: EncryptedDataResponsePayload
Processing report

Processing packet: 1416 bytes input 
Decrypting with provided AES_Key: fa61c523fa61c523fa61c523fa61c523 
Header

LibBroadlink.Common.PacketHeader 
Data Length [56] 
Date 0000-00-00T00:00:00+0000 
Unknown Bytes 10026 
Count 1235 
PacketType Data_Reply (1006|0x3EE) 
ErrorCode Success 
DeviceMac b4:43:0d:f1:df:27 
IPAddress:Port 0.0.0.0:0 
Full Checksum 27671 
Client Id 3 
Payload Checksum 48816 
Header Raw Bytes

5aa5aa555aa5aa55 0000000000000000 
0000000000000000 0000000000000000 
176c00002a27ee03 d30427dff10d43b4 
03000000b0be0000
Payload

LibBroadlink.Common.Responses.EncryptedDataResponsePayload
Data Length [1360] 
Checksum 48816 
Encrypted False 
Encrypted on the wire True 
Payload Raw Bytes

0100000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000 
0000000000000000 0000000000000000

felipediel, 2021-02-18:
Looks like there was no issue. I confirm that the packet can be decrypted with the provided key.

@pavram This onlinebroadlinkinspector is really cool and could have saved me a lot of work if I had known about it before. I like the names you chose for the classes. Very clean. Congrats. The unknown bytes are the PID (devtype).

closed 2021-02-18 by felipediel