mjg59/python-broadlink #96: How to get status Dooya DT360E

Maxmudjon, 2017-05-12:
How to get status Dooya DT360E

mjg59, 2017-05-13:
I don't know, I'm afraid. Do you know if they use the same protocol?

jaxer, 2017-05-20:
it seems so. just got mine here.
```
In [17]: devices = broadlink.discover(timeout=5)

In [18]: devices[0]
Out[18]: <broadlink.device instance at 0x1107cafc8>

In [19]: devices[0].auth()
Out[19]: True

In [20]: devices[0].type
Out[20]: 'Unknown'
```

jaxer, 2017-05-21:
after some digging i can only say that its id is `0x4E4D`. seems to be broadlink protocol, but might be some new flavor of it.

jaxer, 2017-12-24:
opened PR to add Dooya: https://github.com/mjg59/python-broadlink/pull/134

Hoanthieu, 2018-11-20:
@jaxer I am writing code to control dooya cover in HomeAssistant using broadlink.dooya class but when I call open() function, i failed and get a bug like:

File "/srv/homeassistant/lib/python3.5/site-packages/broadlink/__init__.py", line 837, in _send
    return ord(payload[4])
TypeError: ord() expected string of length 1, but int found

This is my code https://github.com/Hoanthieu/Dooya_Controller/blob/master/dooya_cover.py

jaxer, 2018-11-20:
@Hoanthieu right, need to remove `ord` wrapper from `_send` in broadlink. I guess i need to open PR for that. Maybe you can do that?

Thanks for creating proper hass component.

Hoanthieu, 2018-11-20:
@jaxer I'm not sure that I can do that =))). I'm only a beginner

Hoanthieu, 2018-11-20:
@jaxer I tried and could open and stop dooya curtain, but how to get its state, on or off ???

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/134. Thank you!

closed 2021-02-17 by felipediel