mjg59/python-broadlink #97: lpad fix broken?

jdegraef, 2017-05-17:
with the latest version i get this error (on LEDE with python3-minimal)
`Traceback (most recent call last):
  File "./pytest", line 6, in <module>
    device.auth()
  File "/usr/lib/python3.6/site-packages/broadlink/__init__.py", line 203, in au                                                     th
    response = self.send_packet(0x65, payload)
  File "/usr/lib/python3.6/site-packages/broadlink/__init__.py", line 251, in se                                                       nd_packet
    payload=payload.ljust(numpad,"\x00")
TypeError: ljust() argument 2 must be a byte string of length 1, not str
root@zolder:~# ./pytest
Traceback (most recent call last):
  File "./pytest", line 6, in <module>
    device.auth()
  File "/usr/lib/python3.6/site-packages/broadlink/__init__.py", line 203, in auth
    response = self.send_packet(0x65, payload)
  File "/usr/lib/python3.6/site-packages/broadlink/__init__.py", line 251, in send_packet
    payload=payload.ljust(numpad,"\x00")
TypeError: ljust() argument 2 must be a byte string of length 1, not str
`

simple pytest script to trigger error: 
`#!/usr/bin/python3
import broadlink
device = broadlink.a1(("10.0.0.39",80),bytearray.fromhex("aa bb cc dd ee ff"))
device.auth()
`

any suggestion on what's wrong? thanks!

mjg59, 2017-06-05:
Can you try deleting this chunk:

    # pad the payload for AES encryption
    if len(payload)>0:
      numpad=(len(payload)//16+1)*16
      payload=payload.ljust(numpad,"\x00")

and see if that fixes it?

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/108. Thank you!

closed 2021-02-17 by felipediel