mjg59/python-broadlink #99: Cloud API

dsshap, 2017-05-19:
In the readme it says it is not supported. 

Is there a reason why it is not?

Has anyone tried to make it work?

mjg59, 2017-06-05:
No reason, other than that nobody's done the necessary work.

Mucilon, 2017-09-18:
I was thinking of using an arduino to control my cable tv, along with google home (google assistant) on raspberry pi using itfff. But there is this api, very interesting. problem is that there is no time to create the whole system.

jasongi, 2017-09-25:
I've currently got a hacked together project that is a flask API that sends off commands to the broadlink so that I can use it with IFTTT. A lot of things are hard-coded so it's not really shareable but if you look up flask you should be able to get it to work, it's nothing special.

wwarlock, 2018-04-15:
@jasongi May you describe how are you made it? Maybe you have the article already?
Or may you say the names of your ifttt-widgets?

jasongi, 2018-04-15:
@wwarlock I'll try write something up over the next couple of days and upload the code if I have time.

torinnguyen, 2018-04-15:
@jasongi  I'm more interested in your system architecture. Are you able to communicate directly to the Broadlink devices from a cloud server?

jasongi, 2018-04-15:
I have an web server (using nginx/flask) exposed to the internet on my local network that my broadlink is on. It sends the commands to the broadlink using python-broadlink when it receives POST requires from the webhooks action on IFTTT. I'm not using broadlink cloud at all.

torinnguyen, 2018-04-15:
The issue being discussed here is using Broadlink Cloud API directly. Your system architecture relies on python-broadlink locally, that's out of question.

felipediel, 2021-03-12:
In case anyone wants to work on it, I found an interesting project that partially supports the Cloud API. It is written in PHP and translating it into Python is a trivial task, but it requires time that, unfortunately, I no longer have available :(

https://github.com/nick7zmail/MajorDoMo-dev_broadlink