mjg59/python-broadlink #101: how to decode the protocol?

xhan, 2017-05-22:
I currently working on another WIFI-based Air Condition.  
And the official's app which to control it is using an encrypted http protocol. 

broadlink using it's own encrypted protocol too.
can you share some resources or tutorials on how to decode protocol?

Stevenyu1982, 2017-06-01:
which Air condition that you are working on, I have bought the AUX air condition, it use the broadlink_OEM , be able to use the lib to show the deivce. but it says unknow type. will share more informaton, once get home. ,hope we can work together to get the broadlink_oem based air condition work as well.

merlokk, 2017-08-19:
sample for Samurai)
https://github.com/merlokk/BlackBeanControl/blob/master/nec.py

mob41, 2017-08-19:
You have to reverse-engineer the official app to get the encryption key. And use a packet sniffer like Wireshark and specify the decryption method and keys in order to decrypt the packet.

In my opinion, it is not easy as you think. Reverse-engineering really requires you understand the programming language/Assembly very much.

ptd006, 2017-11-08:
Have you tried decompiling the Java code in the APK?  It is possible that everything is in Java.  Even if most code is in a native .so library you may still quickly learn something from the Java code.

merlokk, 2017-11-08:
Dont needs. here https://github.com/mjg59/python-broadlink/blob/master/broadlink/__init__.py#L143

ptd006, 2017-11-08:
@merlokk OP is looking at a different, non-broadlink device (from my reading of the post)