mjg59/python-broadlink #107: Error Connecting to Broadlink device

tobbenb, 2017-06-12:
I can't seem to connect to my RM Pro either through discovery or specifying the IP and MAC.
This is the error I get:

```
2017-06-12 09:47:28.631 User: Admin initiated a switch command (1/Broadlink - Discover/On)
2017-06-12 09:47:28.669 (Broadlink) Calling message handler 'onCommand'.
2017-06-12 09:47:28.670 (Broadlink) onCommand called for Unit 1: Parameter 'On', Level: 0 , Connected : False
2017-06-12 09:47:28.670 (Broadlink) All plugin system is on pause for 5s...
2017-06-12 09:47:33.676 (Broadlink) Found 1 broadlink devices
2017-06-12 09:47:33.676 Error: (Broadlink) 'onCommand' failed 'TypeError':'ljust() argument 2 must be a byte string of length 1, not str'.
2017-06-12 09:47:33.676 Error: (Broadlink) ----> Line 202 in /config/plugins/BroadlinkRM2/plugin.py, function onCommand
2017-06-12 09:47:33.676 Error: (Broadlink) ----> Line 388 in /config/plugins/BroadlinkRM2/plugin.py, function Discover
2017-06-12 09:47:33.676 Error: (Broadlink) ----> Line 203 in /usr/lib/python3.6/broadlink/__init__.py, function auth
2017-06-12 09:47:33.676 Error: (Broadlink) ----> Line 251 in /usr/lib/python3.6/broadlink/__init__.py, function send_packet
```

The plugin is cloned from git and installed. I followed the guide on the domoticz wiki for installing.
I'm using Alpine Linux, Domoticz 3.7712 and Python 3.6.1.
I can't use the pip3 install method as it fails the install with a syntax error that is fixed in the github version.

mob41, 2017-06-12:
Can you specify what is your command used to produce this error?

tobbenb, 2017-06-12:
Sorry. Forgot to tell you how it's produced.
This is what I get in domoticz after I have added Broadlink under Hardware. No matter if I use the default 127.0.0.1 and turn discovery on under switches or use the real IP.

mob41, 2017-06-12:
Take a look of this issue #97. Something most likely to be broken in the library?

tobbenb, 2017-06-12:
The solution in #97 worked.
I did look through the issue list, but didn't see Ipad fix as relevant to the problem I had. 
Is this something that you must fix or an external library?

tobbenb, 2017-06-12:
I thought you were the author @mob41, but see now that you're not :) 
Thanks for the fix. Hopefully the PR gets accepted.

tobbenb, 2017-06-18:
The fix worked here, so closing the issue.

closed 2017-06-18 by tobbenb